Klipper3d/klipper #1830: Slicer configuration - Pretend "Jerk" value for approximating Klipper's motion?

dstulken, 2019-07-23:
So I have a Klipper-related question from the opposite side...

We know that Klipper doesn't have or need configurable "jerk" parameters, and that if a slicer generates G-code which attempts to configure X/Y/Z/E jerk, these commands will be ignored.  The advice given (such as issue #1130), is to not worry about it and not let the slicer do so either.

However, some slicers (such as Cura) still do maintain an internal configuration set of what it "thinks" your printer's settings are.  (Example: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json#L565)
These stored jerk values in the machine profile, along with acceleration parameters, are used for other purposes such as generating print time estimates.  

For this reason, having Cura know Klipper's "jerk" setting will help improve the accuracy of Cura's print time estimates.  But again, Klipper doesn't have a jerk value.   

Acknowledging that this will never be perfect, is there a recommended jerk value to configure in the slicer, that would better approximate the resulting movements generated by Klipper's motion planner?

KevinOConnor, 2019-07-24:
Closest thing to "jerk" would be square_corner_velocity.

-Kevin

KevinOConnor, 2019-08-05:
I'm going to close this issue as it looks like the original question was answered (the closest equivalent is likely square_corner_velocity).

-Kevin

closed 2019-08-05 by KevinOConnor

JohnEdwa, 2019-08-08:
IIRC a Jerk of 10 on a 90deg corner means the printer linearly slows down to 10mm/s, then slams it to 0 and at the same time immediately kicks the other axis to 10mm/s, so the "speed" or time saved is however long it would have taken to do those two skipped 10mm/s accelerations. 

What does a square corner velocity of 10 represent compared to this, as it's clearly not doing the same because of how smoothly it can travel the corner? And obviously it will be completely different if the corner is 45deg or 120 deg for example, but, roughly speaking, any idea?

KevinOConnor, 2019-08-08:
> IIRC a Jerk of 10 on a 90deg corner means the printer linearly slows down to 10mm/s, then slams it to 0 and at the same time immediately kicks the other axis to 10mm/s, so the "speed" or time saved is however long it would have taken to do those two skipped 10mm/s accelerations.

The above is also what Klipper does with a square_corner_velocity=10.

Square_corner_velocity and "jerk" differ slightly on angles other than 90 degrees.

Square_corner_velocity is notably different from "jerk" when the angle doesn't line up exactly with the stepper axes.  For example, "jerk" behaves vastly differently on a corner from 0 to 90 degrees than it does on a corner from 45 to 135 degrees.  In contrast, Klipper performs the same motions on all 90 degree angles.

-Kevin

JohnEdwa, 2019-08-08:
Hmmm, the difference has to indeed be in all those other angles because I know a Jerk of 20 will shake an Ender 3 apart and cause layer shifts (the suggested max Jerk is actually only 8), but a square corner velocity of 20 is perfecly smooth sailing.

If I had to guess it's what happens when the corner is larger, because when the direction changes 180 degrees Jerk will go from +10mm/s to -10mm/s (or +5mm/s to -5mm/s, I'm not sure) instantly, which I assume square corner will not do?

KevinOConnor, 2019-08-08:
Ah, good point.  On a 180 degree angle, square_corner_velocity would slow the head all the way to zero velocity during cornering.  I'm not sure what "jerk" would do in that case - I suspect different firmware do different things.

-Kevin

dstulken, 2019-08-12:
Thanks everyone.   I did an experiment adjusting Cura's assumed jerk value to match Klipper's square_corner_velocity.  (Cura defaulted to 20, my Klipper configuration was at it's default value of 5, so I changed Cura's config to 5 to match.)

I've only timed two prints so far as comparisons, but this did in fact noticeably improve the accuracy of Cura's time estimates, particularly on a longer and more detailed print.   I was having issues with Cura consistently under-estimating the print time by a fair amount, and now with the change it is quite close!   So even though Jerk and square_corner_velocity differ in their implementations on a per-move basis, perhaps some of these differences average out over the duration of a print?   In the big picture, setting them to the same empirically seems to be a good choice.

Kevin, would this be a good addition to the "Slicers" tips in the documentation folder?
https://github.com/KevinOConnor/klipper/blob/master/docs/Slicers.md