Klipper3d/klipper #1831: PROBE_CALIBRATE gives 'Move Out of Range' Unless Z-Offset from last setting increased.

phuzzyday, 2019-07-23:
LOVING Klipper by the way.  I'd actually be a bit lost if I had to go back, I switched early on in my 3D Printing learning!  So I might be biased.

Running stock board on Ender 3 Pro.  Was delighted to find PROBE_CALIBRATE feature. I like to use different Hotend sizes. Using a home made probe hooked to Z Probe line, & servo running off beeper output. it works beautifully.  I suppose I should put on Thingiverse..  

So the problem is as described in title, the PROBE_CALIBRATE command limits the adjustments, hits an error and stops before the head is hitting the paper.  If I manually change the auto generated z-offset to a value known to be above the needed value before I start, I am able to complete the calibration perfectly.  (Using Octoklipper macros to make adjustments.  It's brilliant!)

[klippy.zip](https://github.com/KevinOConnor/klipper/files/3419513/klippy.zip)

FricoRico, 2019-07-23:
For your `[stepper_z]` you need to temporarily set your `position_min` to a sane negative value to stretch the `Out of range` checks. After you've calibrated your probe and set the `z_offset` value, remove the `position_min` or set it to `0` like you have now.

phuzzyday, 2019-07-23:
Ok, thanks.  Still more of a workaround than a fix I suspect, but I'll try it.

FricoRico, 2019-07-23:
I'm not sure if we should consider this a workaround; as it is documented in the [Paper Test](https://www.klipper3d.org/Bed_Level.html#the-paper-test) bed leveling documentation.

It could be made more obvious though, as the first time I also found this to be kind of tucked away beneath a layer of documentation.

KevinOConnor, 2019-08-05:
I'm going to close this issue as it looks like the original question was answered (the position_min should be configured as specified in the documentation).  If you have suggestions on how to better word the documentation then let us know.

-Kevin

closed 2019-08-05 by KevinOConnor