Klipper3d/klipper #1832: [FR] Support for SSD1309 OLED

FricoRico, 2019-07-23:
Hi Kevin,

_Thanks for your great work on Klipper. I really love it so far and got my customized Duplicator 6 up and running in no-time._

However, the display is still not functioning. As said in PR #648 the Duplicator 6 uses an SSD1309 display, which is not supported by the firmware. At the moment the SSD1306 is supported and you pointed out it wouldn't be too hard to support the SSD1309.

I tried finding the differences in controlling a SSD1309 vs SSD1306 but I'm in way over my head. Do you think I can see support for the SSD1309 any time soon?

Thanks for your hard work

KevinOConnor, 2019-07-24:
If you configure it to use the ssd1306 does it work?

-Kevin

FricoRico, 2019-07-24:
Sadly no, I tried that but it didn't. Could also be the wiring config perhaps, but it's hard for me to figure out where to get started without being sure that SSD1309 actually works from the firmware. I use a Anyubic Trigorilla Board in combination with the default Wanhao Duplicator 6 breakout board. It's connected through EXT1 and EXT2 cables.

See this post for the pinouts:
https://github.com/MarlinFirmware/Marlin/issues/8349

KevinOConnor, 2019-07-26:
Alas - it's hard to update the software without knowing if the hardware is mapped correctly.  Do you know what pins the SSD1309 is using?  Is it in spi or i2c mode?

-Kevin

FricoRico, 2019-07-29:
I believe it is SPI connected, but I will try and find out for sure how it's setup by default

FricoRico, 2019-08-16:
Sorry for the late response Kevin, I've found out it's actually connected through I2C. I will confirm that the pinout in my config are set correctly.

KevinOConnor, 2019-08-17:
Ah - looks like you've got a "reset" pin.  If you are up for running test code, I'll try to put together some code to handle that.

-Kevin

KevinOConnor, 2019-08-18:
FYI, I was able to add support for a reset_pin option for SSD1306 displays (commit 96fdb6ef).  You could try pulling the latest code (see the FAQ for details) and update the config with `reset_pin: PE3` in the display section.

-Kevin

FricoRico, 2019-08-18:
I tried the following config (PE3 was for the previous board, the Trigorilla board has more 'regular' pinout)

`[tmc2208 stepper_x]
uart_pin: ar4
microsteps: 16
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 150

[stepper_x]
step_pin: ar54
dir_pin: !ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: ^!ar3
position_endstop: 0
position_max: 200
homing_speed: 50

[tmc2208 stepper_y]
uart_pin: ar5
microsteps: 16
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 150

[stepper_y]
step_pin: ar60
dir_pin: ar61
enable_pin: !ar56
step_distance: .0125
endstop_pin: ^!ar14
position_endstop: 0
position_max: 200
homing_speed: 50

[tmc2208 stepper_z]
uart_pin: ar6
microsteps: 16
run_current: 0.900
hold_current: 0.700
stealthchop_threshold: 30

[stepper_z]
step_pin: ar46
dir_pin: !ar48
enable_pin: !ar62
step_distance: .001258
endstop_pin: probe:z_virtual_endstop
position_endstop: 0
position_min: -5
position_max: 175
homing_speed: 15

[bltouch]
sensor_pin: ar18
control_pin: ar11
x_offset: 9
y_offset: 32
z_offset: 1.625

[bed_mesh]
speed: 120
horizontal_move_z: 5
probe_count: 3,3
min_point: 10,10
max_point: 180,160

[homing_override]
axes: z
gcode:
    G90 ;Use absolute position mode
    G1 Z10 ;Move up 10mm
    G28 X Y
    G1 X100 Y100 F6000 ;Change the X and Y coordinates to the center of your print bed
    G28 Z
set_position_z: 0.0

[extruder]
step_pin: ar26
dir_pin: !ar28
enable_pin: !ar24
step_distance: 0.00248
nozzle_diameter: 0.400
filament_diameter: 1.750
max_extrude_only_distance: 150.0
pressure_advance: 0.150
pressure_advance_lookahead_time: 0.150
heater_pin: ar10
sensor_type: PT100 INA826
sensor_pin: analog13
adc_voltage: 4.96
smooth_time: 0.5
control: pid
pid_Kp: 13.825
pid_Ki: 0.668
pid_Kd: 71.544
min_temp: 0
max_temp: 260

[tmc2208 extruder]
uart_pin: ar42
microsteps: 16
run_current: 1.000
hold_current: 0.750
stealthchop_threshold: 100

[extruder1]
step_pin: ar36
dir_pin: !ar34
enable_pin: !ar30
step_distance: 0.0024
nozzle_diameter: 0.400
filament_diameter: 1.750
max_extrude_only_distance: 150.0
#pressure_advance: 0.200
#pressure_advance_lookahead_time: 0.100
heater_pin: ar45
smooth_time: 0.5
sensor_type: PT100 INA826
sensor_pin: analog14
control: pid
pid_Kp: 13.825
pid_Ki: 0.668
pid_Kd: 71.544
min_temp: 0
max_temp: 260

[tmc2208 extruder1]
uart_pin: ar43
microsteps: 16
run_current: 1.000
hold_current: 0.750
stealthchop_threshold: 100

[heater_bed]
heater_pin: ar8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog15
control: pid
pid_Kp: 64.095903
pid_Ki: 1.649830
pid_Kd: 622.531455
min_temp: 0
max_temp: 110

[controller_fan mainboard_fan]
pin: ar7
max_power: 0.65
kick_start_time: 0.200

[fan]
pin: ar9
kick_start_time: 0.200

[output_pin case_light]
pin: ar44
value: 1

[display]
lcd_type: ssd1306
cs_pin: ar16
dc_pin: ar25
reset_pin: ar27
#spi_bus:
#spi_speed:
#spi_software_sclk_pin: ar23
#spi_software_mosi_pin: ar17
#spi_software_miso_pin:

# Micro-controller information.
[mcu]
serial: /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0
pin_map: arduino

[printer]
kinematics: cartesian
max_velocity: 500
max_accel: 2000
max_z_velocity: 25
max_z_accel: 30

[gcode_macro G10]
gcode:
    G91
    G1 E-2.5 F4000
    G90

[gcode_macro G11]
gcode:
    G91
    G1 E2.5 F500
    G90

# Looking for more options? Check the example-extras.cfg file.`

I measured the reset pin and found it mapped to `ar27` on the new board. This would match the following default RAMPS pinout taken from Marlin firmware:

https://github.com/dot-bob/Marlin-Duplicator-6/blob/bugfix-2.0.x/Marlin/src/pins/ramps/pins_RAMPS.h#L455

However, so far no sign of life. Am I missing something in my config, or might it still be a firmware issue?

KevinOConnor, 2019-08-19:
How did you wire the lcd to the board?  Can you take a photo showing the board, lcd, and their wiring and post it here?

-Kevin

KevinOConnor, 2019-09-06:
Any further updates on this?  FWIW, I'm not sure it is valid to wire an Ultimachine style lcd panel to a trigorilla board (as it seems the ultimatchine style lcd panels require both 5v and 3.3v power).

-Kevin

KeiferMiller, 2019-09-14:
I have a Duplicator 6 and was able to test this.

```
[display]
lcd_type: ssd1306
reset_pin: PE3
```

Popped that into my config and the display started to work.

[image]

KevinOConnor, 2019-09-15:
Okay, thanks.  I updated the example config (commit 661fb613).

-Kevin

KevinOConnor, 2019-09-21:
I'm going to close this issue as it appears basic support is now implemented.

-Kevin

closed 2019-09-21 by KevinOConnor

FricoRico, 2019-09-23:
I'm trying to find more information about the board that is used in the Wanhao Duplicator 6, I suspect it's an Ultimachine LCD panel indeed. But I can't even find any information on that. Have you found a pinout/sheet for the Ultiboard?

uniksoid, 2020-01-18:
I appologize getting back to the closed FR, but I don't wan to create another one which is more like a question.

I have a Re-Arm board running Klipper. When it comes to setting up a display I am facing an issue. 
I have an SPI SSD1306 connected to the board from Marlin set-up. Using the same pins as in Marlin, i.e.:

[display]
lcd_type: ssd1306
cs_pin: P1.9
dc_pin: P2.6 
reset_pin: P0.16

#spi_bus: 0
#spi_speed:

spi_software_sclk_pin: P0.15
spi_software_mosi_pin: P0.18
spi_software_miso_pin:

However I get a Unknown value '' in enumeration 'pin'.
I have read a note in Re-Arm example config saying that only RRD FullGraphic controller is supported with Re-Arm. 

Is it still the case? Or Am I doing something wrong? For example maybe I need somehow specify in config that my display works in SPI mode and not using I2C bus? 

Thanks.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4081793/klippy.log)

jalanjarosz, 2020-01-19:
Just plug in the SSD1306 in the i2c pins, Vcc and GND, then put this in your config:

[display]
lcd_type = ssd1306
i2c_mcu = mcu

It appears the i2c pins are all pre-defined in the micro-controller code.

uniksoid, 2020-01-20:
Thanks for advice. Unfortunately it is not a solution.
I need this display to be connected via SPI. In Marlin 2.0 SSD1306 on Re-Arm works very unstable over i2c, it hangs control board after some time. I guess in Klipper it might be not the case but for some reasons I need my printer to work on both Klipper and Marlin with the same setup.
So any advice on fixing SPI connectivity for SSD1306 is highly appreciated (if possible at all).
Thanks

uniksoid, 2020-01-20:
Removed "SPI_" parameters from config file leaving just cs_pin, dc_pin, reset_pin. 
It is fixed now. The display works perfectly over SPI.

modmt, 2020-08-08:
In my DiY printer i want to use also the SSD1309 over SPI. The SSD1306 works perfect but is too small for me. So i connected the SSD1309 but it did not work. Does anybody know what to change (e.g. in code) to get it to work? Thank you :-)

xiaoqianzi15, 2020-12-01:
reference only BIGTREETECH SKR MINI E3 DIP:
[display]
lcd_type: ssd1306
# SDA: PB7       STM32F103RC
# SCL: PB6
#reset_pin: zboard:PA9
i2c_mcu = mcu