Klipper3d/klipper #1835: 3Dtouch probe working on Anet A8

NorthyIE, 2019-07-24:
Hi,
I have purchased and installed a BLtouch (Full name is Geeetech 3D Touch Sensor High-precision Auto Level Sensor BL touch) on an Anet A8.

When I try homing the printer with the G28-command, the X and Y axis are successfully homed before moving to the center of the print bed. Then the 3D-Touch pin moves down and back up okay (LED goes off and back on as well). Shortly after this, the printer halts and Klipper shows "BLTouch failed to deploy".

I have researched the problem for hours, but cannot find a solution.
I have tried everything as per https://www.klipper3d.org/BLTouch.html#troubleshooting:
BLDEBUG pin_up and pin_down work fine.
"QUERY_PROBE" always show "TRIGGERED", no matter if the pin is up or down. This looks like the issue to me, but I have been unable to fix this.

I triple-checked the cabling which is fine, also the original endswitch is working fine.

I have run the tests as per https://github.com/KevinOConnor/klipper/files/2646233/test-bltouch.gcode.txt and the klipper log is attached.

Thanks for your help :)

[klippy.log](https://github.com/KevinOConnor/klipper/files/3426547/klippy.log)

BlackStump, 2019-07-25:
have you tried 
sensor_pin: ^PC4
as per
https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L17

mariusoei, 2019-07-25:
Is there a second endstop defined for the second z axis? That was the problem for me, the behaviour sounds similar: #1732

NorthyIE, 2019-07-25:
Thanks for your replies.
@BlackStump I have tried with the pull-up flag as well as inverting the pin, but it did not improve the situation unfortunately

@mariusoei I checked, there is no second endstop defined.

I have a feeling that the probe might be defective, so I ordered an original bl touch. Until that arrives, I will keep trying to get this to work in the meantime though :)

NorthyIE, 2019-07-30:
Closing this issue as I have recieved the BL touch and can confirm that the probe is working fine now. The original probe seems to have been defective.

Thanks for your help :)

closed 2019-07-30 by NorthyIE