Klipper3d/klipper #1836: [FR] TMC 2209 slave address (SKR mini E3)

Hywelmartin, 2019-07-24:
As i understand the STM32F103RCT6  is supported, I hope. But the TMC2209 seems to be on the same serial interface and are addressed X=0,Y=1, Z=2 E0=3. 

In the pins file for marlin they all are on serial14
```

#if HAS_DRIVER(TMC2209)
  /**
   * TMC2209 stepper drivers
   * Hardware serial communication ports.
   */
  #define X_HARDWARE_SERIAL  Serial4
  #define Y_HARDWARE_SERIAL  Serial4
  #define Z_HARDWARE_SERIAL  Serial4
  #define E0_HARDWARE_SERIAL Serial4
#endif
```

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/blob/master/firmware/Marlin-2.0.x-SKR-Mini-E3/Marlin/src/pins/pins_BIGTREE_SKR_MINI_E3.h

https://www.mouser.se/datasheet/2/389/stm32f103rc-956261.pdf

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-

KevinOConnor, 2019-07-24:
The tmc2209 supports a uart "address" and it looks like the "skr mini e3" uses that feature.  If someone has the hardware and is willing to run tests, I can probably assist in adding the code.

-Kevin

Hywelmartin, 2019-07-25:
Then I buy one.. see you in a month or so...

/Martin

KevinOConnor, 2019-07-31:
FYI - I updated the code to allow the uart address to be configured (commit a730aec8).  If someone has the board they could try the config at https://github.com/KevinOConnor/klipper/blob/work-mini3-20190731/config/generic-bigtreetech-skr-mini-e3.cfg .

-Kevin

Hywelmartin, 2019-07-31:
So then for all other 2209:s there's no need to ground ms1 or what it was... Just state the the "both pin open" address..

I will test as soon as the board arrive..

pusewicz, 2019-08-22:
@KevinOConnor My board just arrived, I should be able to test this weekend.

looxonline, 2019-08-26:
Also following. Seems like Mike Law sparked some interest in this FW with his new vid.

pusewicz, 2019-08-27:
I did not get to it and now I'm away for the holiday. Gonna have to circle back to it when back home again.

looxonline, 2019-08-27:
My board just landed in the country. Will be about another week until delivered. Will try to test when it arrives.

AshIzat, 2019-08-28:
I have been testing the SKR Mini E3 for a couple of days, for the most part it all works (including uart addressing), but if stepper drivers are not powered with 12/24v before the MCU starts up, then nothing short of a `FIRMWARE_RESET` will get them working properly. This is the issue referenced above..

Hywelmartin, 2019-09-03:
I haven't got my board yet(it just got in to customs today so i might get it next week). but It seems like the feature is there so I'm closing this..

Thanks

closed 2019-09-03 by Hywelmartin