Klipper3d/klipper #1838: Printrbot RevF stepper current correction

Grumps49, 2019-07-25:
There is an error in MCP4728.py. My solution is provided in the attached file. While the code runs, the actual currents are greater than indicated in the printer.cfg file and my board is near the thermal limit. I am aware of the other issues, I believe this is different.
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/3433129/klippy.log.zip)
[Problem with MCP4728 DAC setup for Printrbot Rev F.pdf](https://github.com/KevinOConnor/klipper/files/3433130/Problem.with.MCP4728.DAC.setup.for.Printrbot.Rev.F.pdf)

KevinOConnor, 2019-07-26:
Okay, thanks - commit a878831d.

@drphil3d, @cho934 - FYI.

-Kevin

KevinOConnor, 2019-07-26:
Hi @Grumps49.  FYI, I agree with all of your changes you indicated at https://github.com/KevinOConnor/klipper/issues/1838 and committed them to Klipper (see https://github.com/KevinOConnor/klipper/commit/a878831d5ea8eb86a1bcd94a6a2da926a84d5b6a ).  I think maybe you confused this topic with my unrelated questions at https://github.com/KevinOConnor/klipper/pull/1833 ?

Thanks again.
-Kevin

Grumps49, 2019-07-26:
> Hi @Grumps49 <https://github.com/Grumps49>. FYI, I agree with all of your changes you indicated at #1838 <https://github.com/KevinOConnor/klipper/issues/1838> and committed them to Klipper (see a878831 <https://github.com/KevinOConnor/klipper/commit/a878831d5ea8eb86a1bcd94a6a2da926a84d5b6a> ). I think maybe you confused this topic with my unrelated questions at #1833 <https://github.com/KevinOConnor/klipper/pull/1833> ?

Kevin,

I did indeed get confused. Well this is great! Several people are waiting on me to help them install klipper on their printrbots.

I started with the generic-printrbot.cfg file to set up my printer.cfg file for a Rev F5 board. 

Generic-printrbot.cfg   has [stepper_y] endstop_pin: ^PB0.  Referring to the Rev F4 - F6 schematics, PB0 is the SD card chip select.  ^PB4 is the pin I used. 

That is the only ‘cfg’ vs schematic issue that I found.

Kevin, klipper is an amazing piece of work. Thank you.

Doug Wilson

Grumps49, 2019-07-26:
I looked at #1833 and I had same issues.  My printrbot homes at [0,150], and one pin doesn’t match the schematic for the Ref F5. That was not a bug, just a configuration issue.

Doug

KevinOConnor, 2019-08-05:
I'm going to close this issue as it looks like the original issue is now fixed.

-Kevin

closed 2019-08-05 by KevinOConnor

Grumps49, 2019-08-05:
I would suggest that you update the example-printrbot.cfg file.

Our facebook group has already had one individual that couldn’t print with the old settings and newest driver. The currents are two low.
https://www.facebook.com/photo.php?fbid=10158882279794966&set=gm.2381196435332062&type=3&theater&ifg=1

# Use the following on a Printrboard RevF to control stepper current.
# Needs latest version of mcp4728.py, else currents are not accurate
[mcp4728 stepper_current_dac]
scale: 2.327 
# Marlin full scale is 2.0 amps
channel_a: 1.2 # extruder    	Marlin 60%
channel_b: 1.2 # stepper_z		Marlin 60%
channel_c: 1.0 # stepper_y		Marlin 50%
channel_d: 1.0 # stepper_x		Marlin 50%

I have attached the printer.cfg file that works on my Rev F5. The config file addresses a few things that differ from your example file.

Again, thanks for conceiving of, and then creating Klipper. It looks to me to be a game changer. Good luck in the future.

Sincerely,
Doug Wilson

KevinOConnor, 2019-08-10:
Okay, thanks.  I updated the example config (commit 09f323a0).

-Kevin

Grumps49, 2019-08-10:
Great.  More and more Printrbot users are installing Klipper. This will smooth the process.

Doug Wilson