Klipper3d/klipper #1839: Skr 1.3 + Smart effector not probing

Saso-lab, 2019-07-26:
Since i made all the connection for my delta printer with skr 1.3 by bigtreetech and tmc 2209, the probe of the smart effector does not work when i run the DELTA_CALIBRATE (it crashes on the bed).
I connected it as a normal 3 wire probe to the Z_min on the board and if i touch with the finger the nozzle the led on the smart effector turns on (it means that connections are fine). 
This is the probe's section in printer.cfg:

[probe]
pin: P1.25
x_offset: 0
y_offset: 0
z_offset: -0.1
[image]
[image]

jgill88, 2019-07-27:
Have you tried running the QUERY_ENDSTOPS command while your Z probe is triggered to make sure that it is being read properly?

This might be an issue with the way the inputs for the limit switches are designed on the SKR board, see here: https://reprap.org/forum/read.php?13,855072

I have a NPN inductive probe that is currently working with SKR 1.3 and Klipper. I de-soldered the 10K pull-up resistor on the board (it's R11 for the Z- connection) My inductive probe is working nicely. The processor on the board is 5V tolerant so as long as the signal on the Z-min pin is 0-5V there shouldn't be any issues.

https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/hardware/SKR-V1.3-PIN.pdf

Saso-lab, 2019-07-31:
@jgill88 i queried the endstop while the Z probe is opened and it says triggered, so the logic is not fine. I'm not familiar in soldering things, i've read that the problem can be the 10k pull-up resistor on the board. What can i do? If i try to connect the data line of the sensor to another pin of the board will i have the same issue?

jgill88, 2019-08-01:
I suppose you could try using the Servo pin? it's pin 2.0. It's a digital pin so should be 5V tolerant. 

[image]

[image]

Saso-lab, 2019-08-01:
@jgill88 thanks for the tip, another user also connected the prove at servo pins on the board as you said. But it does not work for me....when i turn the power on the led on the smart effector flashes 7 times (it means that something is wrong) but if i touch it the led detects the contact and turns on.
It is quite strange because i mean if it flashes when i touch it, why it does not change the logical state to perform the delta calibration?

Saso-lab, 2019-08-01:
Ok i try to explain better with some pictures:
1) i connected the probe to servo slot on the board 
2) the led on the effector is off (if i query the endstop to pin 2.0 it says OPEN)
3) if i touch the effector the led turns on, but i queried the pin 2.0 and it did not change the logical state! It was still open! Before and after touching the effector...
[image]
[image]
[image]
[image]

jgill88, 2019-08-01:
What does the probe section of your printer.cfg look like?

jgill88, 2019-08-01:
Also what does your Z stepper section look like?

Saso-lab, 2019-08-01:
This is my main stepper config and the probe config, i remember you that i have a delta printer, so no need of "virtual endstop" if i'm right
```
[probe]
pin: P2.0
x_offset: 0
y_offset: 0
z_offset: 0

[stepper_a]
step_pin: P2.2
dir_pin: !P2.6
enable_pin: !P2.1
step_distance: .0125
endstop_pin: !P1.28
#position_endstop: 520
homing_speed: 50
#arm_length: 360.21

[stepper_b]
step_pin: P0.19
dir_pin: !P0.20
enable_pin: !P2.8
step_distance: .0125
endstop_pin: !P1.26

[stepper_c]
step_pin: P0.22
dir_pin: !P2.11
enable_pin: !P0.21
step_distance: .0125
endstop_pin: !P1.24
```

Saso-lab, 2019-08-01:
Last update, i discovered the command query_probe (the probe changes its state in the right way), but i can't do the delta calibration because there is a new error.... The effector stops before reaching the bed (few centimeters from it)
[image]

closed 2019-08-02 by klipper-gitissuebot