Klipper3d/klipper #1840: How do I install Klipper (RPi 3) on MKS SBase 1.3 board (for Creality CR-10)?

scarecrow2k6, 2019-07-26:
This might have been mentioned numerous times so I apologize in advance. I could not fully find a step by step solution on how to install Klipper on MKS Sbase 1.3 board. I followed the normal route but I get an error when trying to "make flash flash_device= xxx" ( I did put the actual port ). The "make" command did generate a klipper.bin file,  which I read somewhere to manually put it on SD card and rename it firmware.bin and restart the board. I did and got firmware.cur, which I'm guessing is good. So far I have only firmware.cur file in the SD. I matched the config file (CR-10 board) to the MKS as much as I could and made a printer.cfg file. Which I replaced with the old one. Restarted Pi and went to Octoprint and typed "status" in terminal and I get, "Klipper state: Not ready". Any ideas? Thanks.

BlackStump, 2019-07-26:
check your thermistors
this is from your log
MCU 'mcu' shutdown: ADC out of range
This generally occurs when a heater temperature exceeds
its configured min_temp or max_temp.
temp=-99.6 
both extruder and bed are reading wrong

scarecrow2k6, 2019-07-27:
I tried the board without hooking anything up, not even thermistors so that I know it works. Forgot that  for safety FW checks fpr  a bunch of sensors :) .  Hooked the thermistors and error went away!

closed 2019-07-27 by scarecrow2k6