Klipper3d/klipper #1842: Receiving Printer is not ready on SKR V1.3

tgmorris99, 2019-07-27:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3437642/klippy.log)
[OctoPrint.txt](https://github.com/KevinOConnor/klipper/files/3437641/OctoPrint.txt)

I'm trying to bring up Klipper on the SKR V1.3 along with octopi 0.16.0 and having trouble getting the printer to connect.

I built klipper and modified the config file and there are no errors on the initial connection in OctoPrint. However a STATUS command fails with a "Printer is not ready" error.

The usb port is supposed to be /dev/ttyACM0, which works fine if Marlin is booted. However as soon as I switch to klipper (copy klipper.bin to firmware.bin on the SD card and reboot the SKR to create FIRMWARE.CUR) /dev/ttyACM0 no longer appears on the pi - so of course it's going to fail.

I'm probably missing something obvious but I certainly am unable to see it.

Looking the port up by id doesn't work.
```
pi@octopi:~/klipper/out $ ls /dev/serial/by-id/
ls: cannot access '/dev/serial/bi-id/': No such file or directory
```

tgmorris99, 2019-07-27:
Some additional info.

I tested by plugging in my Mega2560 w/Marlin on it and the port immediately shows up. If I had to guess, it looks like the USB port on the SKR isn't active once the klipper FW is loaded on it as the port works fine if Marlin is loaded on the SKR.

KevinOConnor, 2019-07-27:
Lots of people have reported success with Klipper on the SKR v1.3 - I'm not sure what's different with your setup.  I'd try running `make distclean` followed by `make menuconfig`, `make`, and recopy the resulting out/klipper.bin to the SD card.  Select LPC1768, but otherwise leave everything else at the default in make menuconfig.

-Kevin

tgmorris99, 2019-07-27:
Well, that didn't work either so I did klipper-uninstall.sh and redid everything. It's now seems to be working.

Clearly something must have gotten buggered on the earlier install.

closed 2019-07-27 by tgmorris99