Klipper3d/klipper #1844: [FR] Reset temperature fault via gcode without restarting klipper

drphil3d, 2019-07-28:
This is an incredibly useful feature available on the Duet. Say for instance your 40 hours into a 50 hour print and something comes loose causing a heater fault which causes your printer completely shut down, tons of wasted time and plastic. If instead the heater that's at fault shuts down it would save a lot of time and headaches especially when you have more then one print head.
https://duet3d.dozuki.com/Wiki/Gcode#Section_M562_Reset_temperature_fault

https://duet3d.dozuki.com/Wiki/Spurious_heater_faults_and_how_to_avoid_them#Section_How_to_reset_a_heater_fault_without_restarting_the_Duet

KevinOConnor, 2019-07-29:
My high-level feedback is that this sounds useful if someone wishes to implement it.

My advice would be to try and come up with a mechanism that handles error recovery in general - that is, on any error, pause the print and ask for user feedback - then continue or cancel.  I don't think this is something that can effectively be done entirely in Klipper today - it would need more interactive feedback (eg, a web page or octoprint plugin).

-Kevin

drphil3d, 2019-07-29:
Sounds like a good idea. Fault handler that can be enabled and configured to ask for user input instead of immediately going into shutdown mode, much the same way stall guard feature is implemented in duet firmware which pauses the print and waits for input upon detecting skipped steps.

KevinOConnor, 2019-09-06:
I'm going to mark this as inactive as it doesn't appear there is active development on it right now.

-Kevin

closed 2019-09-06 by KevinOConnor