Klipper3d/klipper #1845: How to tune pressure_advance_lookahead_time and square_corner_velocity?

TERRAOperative, 2019-07-29:
I've been tuning a new klipper install on my 3D printer and I am getting ghosting/ringing after corners.

I have been looking through the documentation on how to tune the 'pressure_advance_lookahead_time' and 'square_corner_velocity' parameters, but I can't find anything.

Can anyone please let me know how to tune these parameters properly?
Or if there is another parameter I should be tuning to remove the ringing? It seems to print perfectly otherwise.

TERRAOperative, 2019-07-29:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3440081/klippy.log)

Xono, 2019-07-29:
@TERRAOperative 

The steps for pressure advance tuning are detailed in [docs/Pressure_Advance.md](https://github.com/KevinOConnor/klipper/blob/master/docs/Pressure_Advance.md). This document also contains a note that the `pressure_advance_lookahead_time` value should be set to 0 while tuning, and be returned to its default of 0.010 for normal use - though it can be adjusted manually if needed. The `square_corner_velocity` field is described in [docs/kinematics.md](https://github.com/KevinOConnor/klipper/blob/master/docs/Kinematics.md) as being used to calculate the speed through different sharp angles, though not how different values affect the print itself. 

It is detailed that a restart command will reset both of these to their defaults, meaning you generally shouldn't need to alter these manually - the extruder's `pressure_advance` value is the only one you should need to be tweaking by hand yourself, as described in the steps of the Pressure Advance document.

BlackStump, 2019-07-29:
try reducing your acceleration, especially if it is a bed slinger ie Ender3
your log has 
max_accel = 3000
Try
max_accel = 1000

TERRAOperative, 2019-07-29:
@Xono I have done the Pressure_Advance tuning which worked well for the corners, but I didn't see the kinematics document yet, thanks for the link. :)

@BlackStump I tried a test print at max_accel = 2000 and the ringing is less now, so that is on the right track.
My printer is an extensively modified Wanhao Duplicator 6, so I was hoping it would be less prone to this sort of thing, having a full Ultimaker type enclosure frame.
I wonder if the stepper motor dampers I fitted are causing a little bit of compliance and causing the ringing?...

More tweaking ahoy. :D

KevinOConnor, 2019-08-17:
It sounds like the original question raised in this issue was answered.  Any further updates on this issue?

-Kevin

closed 2019-09-06 by KevinOConnor