Klipper3d/klipper #1849: BLTouch v3 clone requires manually pulling the pin down to start working

rmawatson, 2019-07-29:
I have recently changed my BLTouch to a (clone) version with the plastic pin. The previous one was giving large errors in the position (+/0.5) every so often. I wired in the new one, using the same pins, however when klipper is first started from power on, the log reports

```
Failed to verify BLTouch probe is raised; retrying.
Failed to verify BLTouch probe is raised; retrying.
BLTouch raise probe error: BLTouch failed to raise probe
```
I have tried 
```
BLTOUCH_DEBUG COMMAND=reset
BLTOUCH_DEBUG COMMAND=pin_up
BLTOUCH_DEBUG COMMAND=pin_down
```
None of them appear to work. The blue light does go on though. Reset does appear to reset the probe and the red light goes off.

However, as can bee seen in the linked video, after I pull the pin down manually one time, everything starts working, pin_up, pin_down.

could there some initial condition about this probe that is not what klipper expects that is causing it to not initialize properly?

- The video is from powering up the printer/pi
- up to `1:16` I am trying `pin_up/pin_down` after octopi has started
- At `1:17` I call `BLTOUCH_DEBUG COMMAND=reset` and it resets the probe.
- I then try the `pin_up/pin_down` commands again, but they still do not work.
- After manually pulling the pin down, everything works

[Video](https://photos.app.goo.gl/qhKb3ey1HquYAf7Q6)

Board: MKS-GenL
Pi: Rasberry-Pi V3
Config: Attached in log file

[klippy.log.bltouchproblem.txt](https://github.com/KevinOConnor/klipper/files/3443918/klippy.log.bltouchproblem.txt)
](url)

BlackStump, 2019-07-29:
this is from Antclabs manual
※ If push-pin down error occurs, turn core 180 degrees with Allen-key so that the core is further inside the casing.
This is referring to the center grub screw, worth a try as it is most likely a problem with the device then a issue with the firmware imho.
https://docs.wixstatic.com/ugd/f5a1c8_c2e829cc51d34a529518f3248f3fd883.pdf

rmawatson, 2019-07-29:
I'll take a look at that. But it is not as if it doesn't work atall, once I do a manual pullout of the pin, it will go up and down without issue. Z homes, it does a full Bed Tilt probe with 4 points, 4 samples per point, good tolerance.

Pretty strange.

rmawatson, 2019-07-29:
Great call, thanks. That has indeed fixed it. For the next person who has this, I had to really crank the grub screw down. It didn't feel like its going to move. backing it off half a turn then tightening it down seems to have fixed this. 

Thanks @BlackStump

closed 2019-07-29 by rmawatson