Klipper3d/klipper #1850: Reducing TMC current while homing

cmroche, 2019-07-29:
Recently I've setup Klipper with TMC2130's using sensorless homing. I noticed that the carriage bands into the frame somewhat enthusiastically. On the Duets, it's possible to prevent this in the homing script by setting a current factor, ie: 0.5 to run at 50% nominal current. This bring the noise down, as well ask shaking, and maybe potential for damage when using this feature.

Looking into the extended gcode I've been able to achieve this with something like this:

```
[homing_override]
set_position_z:6
axes: z
gcode:
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT=0.400 HOLDCURRENT=0.400
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT=0.400 HOLDCURRENT=0.400

    ... Home or something ...    

    SET_TMC_CURRENT STEPPER=stepper_x CURRENT=0.850 HOLDCURRENT=0.500
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT=0.850 HOLDCURRENT=0.500
```

Which I suppose works, but should I ever change motors or currents I need to make updates in multiple places which I am gauranteed to forgot.

So the question, does it make sense to have something like this (note homing_current):

```
[tmc2130 stepper_x]
run_current: 0.850
hold_current: 0.500
homing_current: 0.400
driver_SGT: 2
diag1_pin: ^!ar3
sense_resistor: 0.11
```

Then during the homing sequence one could iterate through the TMC drivers, call a homing prepare to setup and restore the desired currents automatically?

If the value is not supplied, the default would be to leave it at the normal run current.

If there is an interest, I am willing to make and submit this change in the next couple of weeks.

cmroche, 2019-08-03:
Well I've started work on this, it won't take long to complete and test though it'll probably be next week before I send a PR.

KevinOConnor, 2019-08-05:
As observed, one can configure the current via SET_TMC_CURRENT and homing_override, but there isn't an automated mechanism for this.  Given the large variance in homing procedures, I'm not sure an automated mechanism will help that much, but if you come up with something then let us know.

-Kevin

cmroche, 2019-08-05:
The objective isn't so much to automate the homing procedure than to give a means that avoids copy and pasting current values in multiple locations. Similar to Duet's `[M913](https://duet3d.dozuki.com/Wiki/Gcode#Section_M913_Set_motor_percentage_of_normal_current)` gcode. 

There is an issue with using SET_TMC_CURRENT though in the `[homing_override]`. Consider the following config:

```
[homing_override]
set_position_z:6
axes: z
gcode:
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT=0.400 HOLDCURRENT=0.400
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT=0.400 HOLDCURRENT=0.400
```

And then doing a `G28 XY`. Looking at the code, my expectation here is that the currents will not be adjusted.

But then consider this configuration to support a BL-Touch

```
[homing_override]
gcode:
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT=0.400 HOLDCURRENT=0.400
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT=0.400 HOLDCURRENT=0.400

    G90 ; Use absolute position mode
    G1 Z10 ; Move up 10mm
    G28 X Y
    G1 X166 Y120 F6000 ; Change the X and Y coordinates to the center of your print bed
    G28 Z

    ...
```

This will run on any combination of `G28`, `G28 X`, `G28 XY` forcing a Z-Probe if it is not desired, and potentially unexpectedly.

Alternatively if the overrides were provided independently, is `homing_override_all`, `homing_override_x`, `homing_override_y`, `homing_override_z` then a parameter might be added to give `SET_TMC_CURRENT STEPPER=stepper_x SCALE=0.5` allowing a highly customizable procedure, though I cannot immeditely think of a case where this is needed.

My thought is that adding an option to the stepper configuration would meet most if not all use cases without impeding on other aspects of homing procedure.

KevinOConnor, 2019-08-05:
> The objective isn't so much to automate the homing procedure than to give a means that avoids copy and pasting current values in multiple locations.

Yes, I understand.  Given all the funky things that are done with homing, I'm not sure one can have a single "homing_current".  But, I'm open to it if you wish to explore it.

-Kevin

KevinOConnor, 2019-09-06:
I'm going to close this issue as it appears the current conversation has concluded.  Feel free to submit a pull request when ready.

-Kevin

closed 2019-09-06 by KevinOConnor

blondak, 2019-09-10:
OK, i wrote a small patch, that can handle homing currents

philjynx, 2021-05-08:
I know this in ancient thread but I have wasted some time (days) recently because I forgot to change the current in one area of my config.
My workaround to have the currents 'configured' and maintained in just one place is this:
I now run those macros as and when I need to, knowing that the outcome is reliable.
I'm running FIRMWARE_VERSION:v0.8.0-549-gb80d2f77 
[gcode_macro Set_Home_Current]
gcode:
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT=0.600 HOLDCURRENT=0.300
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT=0.600 HOLDCURRENT=0.300
	
[gcode_macro Set_Print_Current]
gcode:	
  SET_TMC_CURRENT STEPPER=stepper_x CURRENT=1.414 HOLDCURRENT=1.414 
  SET_TMC_CURRENT STEPPER=stepper_y CURRENT=1.414 HOLDCURRENT=1.414