Klipper3d/klipper #1851: TCP connections over pty (for dockerizing)

mboyer85, 2019-07-30:
Hi, this is for an enhancement, why not integrate a listening socket over the pty like there https://stackoverflow.com/questions/48781155/how-to-connect-inet-socket-to-pty-device-in-python
Docker is a one processus launching (vs adding socat).
Thanks for this great tool(I use klipper on my Haribo and on my voron 2.1)

mboyer85, 2019-07-31:
Docker is a one processus launching : I mean one container/ one processus

vladbabii, 2019-08-01:
You can also use socat and ser2net (on same host and container or remote host and container) to "attach" serial connections via tcp/ip to any virtual/physical device.

mboyer85, 2019-08-01:
Socat or ser2net is one processus, Klipper another, they need to be managed by a script that spawn the two processus in the same container (not a best practice in Docker), moreover open a TCP port directly in python is not so complex...

mboyer85, 2019-08-01:
The objective : one container for octoprint, one container for the bridge(socat) and one container for Klipper. The whole managed by Docker stack for x printers, octoprint can be deployed clustered and linked to its Klipper instance, all that containers with volumes for config files (provided by a nas)

mboyer85, 2019-08-01:
Thus all can be updated independently, all that things  work on raspberry and the like, they can  be scaled more quickly with Docker (Klipper instances stay near printers)

mboyer85, 2019-08-01:
All the Docker pros.... Constrained to our needs for multi printers

mboyer85, 2019-08-01:
All that things seem overkill but think deploy vs install (on each instances) Can be a +++

vladbabii, 2019-08-03:
This dockerfile runs 2 processes - socat and home assistant. you could change the one running homeassistant to manage klipper - https://github.com/vladbabii/homeassistant-socat

mboyer85, 2019-08-03:
Thanks that is whath I will do if Kevin do not add the TCP listening directly in python

KevinOConnor, 2019-08-05:
This sounds similar to issue #1097.  If someone wishes to implement it, then that's great.

-Kevin

vladbabii, 2019-08-14:
@KevinOConnor how much latency between klipper process and mcu is ok? 

I could run ser2net on a very small router to move the usb serial port of the mc to a tcp connection and use a server across the world to run klipper and controll that printer. For example a mr3020 with openwrt is pretty small - https://i.ytimg.com/vi/1GWBR_ONI6Y/maxresdefault.jpg

For multiple printer i would add a mr3020 to each ( or a hub to mr3020 and a 4-6 printers per mr) and run klipper dockerized on a single server.

I want to try this but I'm not sure what would be the maximum delay between the socat and ser2net endpoints. Is under 5 miliseconds ok? Or do I need under 1 ms ?

KevinOConnor, 2019-08-17:
> how much latency between klipper process and mcu is ok?

That's not something I've tested.  As a guess, I'd say keep it under 10ms.

-Kevin

vladbabii, 2019-08-20:
I've started work on a klipper + octoprint docker stack with separate containers. Will post updates here when I have a working solution. From my limited testing socat and ser2net work fine

vladbabii, 2019-08-21:
I'm almost done, just can't get octoprint to play nice with socat.

Docker container
(A) takes a serial port and maps it over network with ser2net
(B) runs ser2net to grab port from A, run klipper on that port, and take klipper serial and output to ser2net
(C) octoprint runs socat to get klipper serial port and uses it

I'm stuck at (C), klipper runs fine with the whole stack on my laptop, using (A) with the serial port from a skr mini attached via usb

KevinOConnor, 2019-09-14:
I'm going to close this as it looks like the conversation has concluded.  If there's an interest in adding this, please open a pull request with the support.

-Kevin

closed 2019-09-14 by KevinOConnor