Klipper3d/klipper #1852: Firmware retraction - MCU shutdown: Rescheduled timer in the past

1walkerit, 2019-07-30:
Hi,
I am just testing FW retraction using new slicer. At this time I do not realy know where is the proble- if slicer or klipper side.
I've tried to print using FW retraction but got the error MCU shutdown: Rescheduled timer in the past after first layer.
When I tried the same mode, same settings but normal retraction  - all is OK. So can you help to find problem - or bug?
Thank you

KevinOConnor, 2019-08-05:
The error message should have shown up in the OctoPrint terminal:
```
MCU 'mcu' shutdown: Rescheduled timer in the past
This generally occurs when the micro-controller has been
requested to step at a rate higher than it is capable of
obtaining.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
```

You've requested more steps per second than the micro-controller can support.  The issue is likely related to pressure advance - a similar report is at #865.

-Kevin

1walkerit, 2019-08-12:
In this case I do not understand why with slicer's retraction it is woriknig and with FW retraction not. The values are the same (retraction speed, lenght and PA). Does it mean if I want to use FW retraction I have to disable PA?

KevinOConnor, 2019-08-17:
Make sure you're using the correct units - Klipper is mm/s while other software sometimes use the (very confusing) mm/minute.  Other than that, I also don't know why they would be different.

-Kevin

KevinOConnor, 2019-09-06:
I'm going to close this issue as it appears the conversation has concluded.

-Kevin

closed 2019-09-06 by KevinOConnor