Klipper3d/klipper #1854: [FR] TMC2130 homing current

blondak, 2019-08-01:
Hi, is possible do something like homing current for TMC2130 sensorless homing?

I'm running relatively high current on HyperCube and I'm using sensorless homing, i prefere faster movement, so i use run_current: 1 but even with driver_SGT: -64 it takes long time before driver detects stall, when I use run_current: 0.5 then everthing goes perfect. 

Is there posibility create etc MACRO which change run_current before G28 and then goes back?

Hywelmartin, 2019-08-01:
The "gcod" will alter the current
SET_TMC_CURRENT STEPPER=<name> CURRENT=<amps> HOLDCURRENT=<amps>

use the homing override to alter the homing procedure 
https://github.com/KevinOConnor/klipper/blob/c356e8362dc555ced5a137a1aabc73e95cf29561/config/example-extras.cfg#L1295

blondak, 2019-08-01:
thank you, this is what I was looking for ;)

blondak, 2019-08-01:
OK, i made a small patch, that allows using config keyword in CURRENT and HOLDCURRENT

diff --git a/klippy/extras/tmc2130.py b/klippy/extras/tmc2130.py
index e6a27a4..35f024c 100644
--- a/klippy/extras/tmc2130.py
+++ b/klippy/extras/tmc2130.py
@@ -99,12 +99,12 @@
         self.name = config.get_name().split()[-1]
         self.mcu_tmc = mcu_tmc
         self.fields = mcu_tmc.get_fields()
-        run_current = config.getfloat('run_current',
+        self.default_run_current = config.getfloat('run_current',
                                       above=0., maxval=MAX_CURRENT)
-        hold_current = config.getfloat('hold_current', run_current,
+        self.default_hold_current = config.getfloat('hold_current', self.default_run_current,
                                        above=0., maxval=MAX_CURRENT)
         self.sense_resistor = config.getfloat('sense_resistor', 0.110, above=0.)
-        vsense, irun, ihold = self._calc_current(run_current, hold_current)
+        vsense, irun, ihold = self._calc_current(self.default_run_current, self.default_hold_current)
         self.fields.set_field("vsense", vsense)
         self.fields.set_field("IHOLD", ihold)
         self.fields.set_field("IRUN", irun)
@@ -141,13 +141,19 @@
     def cmd_SET_TMC_CURRENT(self, params):
         gcode = self.printer.lookup_object('gcode')
         if 'HOLDCURRENT' in params:
-            hold_current = gcode.get_float(
-                'HOLDCURRENT', params, above=0., maxval=MAX_CURRENT)
+            if gcode.get_str('HOLDCURRENT', params).lower() == 'config':
+                hold_current = self.default_hold_current
+            else:
+                hold_current = gcode.get_float(
+                    'HOLDCURRENT', params, above=0., maxval=MAX_CURRENT)
         else:
             hold_current = self._calc_current_from_field("IHOLD")
         if 'CURRENT' in params:
-            run_current = gcode.get_float(
-                'CURRENT', params, minval=hold_current, maxval=MAX_CURRENT)
+            if gcode.get_str('CURRENT', params).lower() == 'config':
+                run_current = self.default_run_current
+            else:
+                run_current = gcode.get_float(
+                    'CURRENT', params, minval=hold_current, maxval=MAX_CURRENT)
         else:
             run_current = self._calc_current_from_field("IRUN")
         if 'HOLDCURRENT' not in params and 'CURRENT' not in params:

Hywelmartin, 2019-08-01:
#1850

blondak, 2019-08-01:
> #1850

I'm sorry, I didn't notice :disappointed:

KevinOConnor, 2019-08-17:
Thanks.  Lets close this in favour of the existing issue #1850.

-Kevin

closed 2019-08-17 by KevinOConnor