Klipper3d/klipper #1855: "Probe" command fails with BL Touch.

nearlyNonexistent, 2019-08-02:
https://termbin.com/2q3d - /tmp/klippy.log
Hardware: legitimate BL touch v2 installed on an Ender 3.

Installed a BLTouch with the recommended config for Ender 3, trying to work with examples from previous issues; did I do something wrong? Using a pin 27 adapter.

nearlyNonexistent, 2019-08-02:
Oh, it has to be an actual attachment for the bot to find it. Here.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3459592/klippy.log)

JohnEdwa, 2019-08-03:
>Hardware: legitimate BL touch v2 installed on an Ender 3.

>klippy.log:
[bltouch]
pin_up_touch_mode_reports_triggered = False

```
#pin_up_touch_mode_reports_triggered: True
#   Set if the BLTouch consistently reports a "triggered" state after
#   the commands "pin_up" followed by "touch_mode". This should be
#   True for a genuine BLTouch v2 and earlier; the BLTouch v3 and some
#   BLTouch clones require False. The default is True.
```

Could be that.

KevinOConnor, 2019-08-05:
If you have a genuine bltouch, then the configured sensor pin is likely not correct.  Be sure to follow the directions at https://www.klipper3d.org/BLTouch.html

-Kevin

KevinOConnor, 2019-09-14:
I'm not sure what the status of this issue is.  There's been some discussion in issue #1938 that may be related.  However, the description sounds more like a configuration issue than a software issue.  The log does not show the reported "Failed to verify BLTouch probe is raised; retrying" error.

-Kevin

closed 2019-09-29 by KevinOConnor