Klipper3d/klipper #1856: Menu items for movemment have bed size hard coded

tgmorris99, 2019-08-03:
Log file really isn't needed for this one.

When using the LCD display the motion control items have hard-coded values for the bed size instead of using the values from the printer.cfg file - if that's even possible.

Here's a snippet from the current version of menu.cfg

```
[menu __control __move_10mm __axis_x]
type: input
name: "X:{0:05.1f} "
parameter: gcode.move_xpos
input_min: 0
input_max: 200.0
input_step: 10.0
gcode:
    G90
    G1 X{0:.1f}
```
It would be nice if the input_min/input_max values could reference **_stepper_x.position_min / stepper_x.position_max_** values so that users with larger build volumes don't need to edit the menu.cfg file before doing the make. Same goes for the Y & Z axis as well - though for Z the min should probably refer to the endstop.

KevinOConnor, 2019-08-05:
FYI, the menu is in a state of flux right now as we convert them to use "jinja2 templates" (PR #1518).

-Kevin

KevinOConnor, 2020-08-18:
FYI, with the merge of the new menu v2 code (#2344), this should be easier to fix.

-Kevin

closed 2020-09-15 by github-actions[bot]