Klipper3d/klipper #1857: MKS Sbase 1.3 with LCD 128x64?

scarecrow2k6, 2019-08-03:
Trying to use
[image]

This is similar to: [Issue 612](https://github.com/KevinOConnor/klipper/issues/612) except this LCD has 1 plug (EXP3).

Using this code:
[display]
lcd_type: st7920
cs_pin: P0.16
sclk_pin: P0.15
sid_pin: P0.18
encoder_pins: ^P3.25, ^P3.26
click_pin: ^!P1.30

Trying to use CR-10 with a MKS Sbase 1.3. The CR-10 came with a 128x64 lcd (pic attached). The thing is, it has only one plug (EXP3) which worked perfectly with the Melzi board which came with the CR-10. How do I use it with the MKS Sbase 1.3? Most of the things I could find online are of 2 plugs (EXP 1 and EXP 2). 

Just wanna confirm that I have a RepRap Discount LCD 128x64 (2 plugs flipped) and the above code works perfectly well but tried using the One Plug lcd on EXP1 flipped but does not work.

scarecrow2k6, 2019-08-04:
Anyone?

KevinOConnor, 2019-08-05:
You'll need to find the schematic for your board, find which pins are mapped to the EXP1 header, find the pins on the CR-10 display, and then map the equivalent pin to the equivalent function in the config file.

If you can obtain the correct mapping, we can add an example to the Klipper config file for future users.

-Kevin

KevinOConnor, 2019-08-17:
FYI - it doesn't appear one can connect the single EXP1 plug directly to the SBase, as the SBase doesn't route all the pins on that connector.  It may be possible to use jumper wires (or similar) to make a custom connection between lcd and board.

-Kevin

KevinOConnor, 2019-09-06:
I'm going to close this issue as it appears the conversation has concluded.

-Kevin

closed 2019-09-06 by KevinOConnor