Klipper3d/klipper #1858: Using gcode_button my_gcode_button results in Failed to verify BLTouch probe is raised; retrying.

vladbabii, 2019-08-03:
If i add to my config

```
[gcode_button emerg_stop]
pin = ar15
press_gcode = 
	M112
release_gcode = 
	FIRMWARE_RESTART
```

Then i get the error

```
Failed to verify BLTouch probe is raised; retrying.
Failed to verify BLTouch probe is raised; retrying.
```

If i comment the above config lines everything works normally.

vladbabii, 2019-08-03:
[klipper.zip](https://github.com/KevinOConnor/klipper/files/3463736/klipper.zip)

Log file of the issue

KevinOConnor, 2019-08-21:
The log indicates that the ar15 pin is being reported as pressed and is thus running the M112 command.  The bltouch message is a bit confusing, but the underlying reason should be reported via the STATUS command.

You may need a pullup and/or an invert on the pin for it to work correctly.

-Kevin

KevinOConnor, 2019-09-06:
I'm going to close this issue as it appears the original question was answered (the shutdown was due to the button registering as being pressed).

-Kevin

closed 2019-09-06 by KevinOConnor