Klipper3d/klipper #1859: fan_soft_pwm marlin feature missing

maroonds, 2019-08-03:


FHeilmann, 2019-08-05:
What exactly are you having issues with? Klipper soft-pwms all pin by default unless you specifically declare the pin as a hardware_pwm pin (by configuring it as `hardware_pwm: True`).

maroonds, 2019-08-05:
I have to use a buck converter to get my 24v output to work with my 5v fan,
because of that I’m unable to change the fan speed since it’s always
outputting 5v.
With the marlin feature fan_soft_pwm I was able to still change the fan
speed.
It would be really cool if you could implement this feature to Klipper.:)

dragonnn, 2019-08-05:
You are using you step-down converter wrong.
Connect it directly to the power supply and connect the fan positive wire to the step-down converter and next connect the fan negative wire to the fan PWM output of you board.
This is the proper way of using a step-down converter for driving fans with PWM control and in this way you don't need any "slow pwm" o other hacks

maroonds, 2019-08-05:
Okay thanks! Will this also work with a 24v psu?

dragonnn, 2019-08-05:
Yes, with any PSU this will work as long you step-down module can handle such input voltage (and almost any can accept 24V)

maroonds, 2019-08-05:
Okay thanks! But why can I adjust the fan speed with marlin?

closed 2019-08-11 by klipper-gitissuebot

maroonds, 2019-08-17:
Hi! Do you know where the fan PWM output is on the Mks gen 1.4?