Klipper3d/klipper #1860: Macro help request: how to check if printer is homed, and if BED_MESH_CALIBRATE has been done?

JohnEdwa, 2019-08-04:
TL;DR: how do I query in a G-code macro if:

* The printer has been homed or not.
* If the bed calibration has been done once this session?

I would like to use a 5x5 grid for the BLtouch, but doing it before every single print is kinda excessive, while saving it permanently isn't really optimal either. I'd like to create a macro that would check if the calibration data exists or not, so it would only do it on the first print of the session and making it redo it would be as simple as rebooting the printer before the next print.

---

(Don't mind me, just checking if I can keep the bot satisfied. [klippy.log](https://github.com/KevinOConnor/klipper/files/3465309/klippy.log) )

KevinOConnor, 2019-08-05:
Neither setting is currently exported right now.  It would be possible for someone to add `get_status()` methods to export the information so it would be available from "gcode templates".

-Kevin

JohnEdwa, 2019-08-08:
It seems PR #1866 adds the homing check, so now only the bed calibration check is missing.

Though it might be even more useful if there was a way to do it from the G-code command itself. Not sure on the best parameter name, but as an example, adding in``BED_MESH_PROFILE USE=temp`` that will load a profile named "temp" if it exists, or automatically run BED_MESH_CALIBRATE to generate it if it doesn't.
Related to that, it would also be useful if there was a way to query all the valid profiles (either by just ``BED_MESH_PROFILE`` or something like ``BED_MESH_PROFILE LIST``.

KevinOConnor, 2019-09-21:
I'm going to close this issue as it appears the conversation has concluded.  If additional information is to be exported it will require a developer to submit a PR that exports the information.

-Kevin

closed 2019-09-21 by KevinOConnor