Klipper3d/klipper #1862: Mesh bed leveling offsets aren't working

daiben2, 2019-08-05:
I configured the mesh leveling with an ezabl and the offsets aren't working. It tries to probe the 20,20 point with the nozzle instead of the probe. The nozzle crashes into the bed at the 20,20 position while the probe is off the bed. Also the Z height nor works it prints into the air even with a -4.50mm offset it prints at the same height no matter what I do

[klippy.log](https://github.com/KevinOConnor/klipper/files/3469048/klippy.log)

Arksine, 2019-08-06:
In Klipper all probed points (or in the case of bed_mesh the min_point and max_point) refer to nozzle position.  You need to make sure you don't use points that move the probe off of the bed.  The x and y offsets are applied to the mesh when its generated after the probing procedure.

Setting the z offset to a negative number would actually raise the probe.  Its recommended to use PROBE_CALIBRATE to determine your z offset.

daiben2, 2019-08-06:
And where can I check the mesh because in the terminal window the offsets aren't applied

mental405, 2019-08-19:
[In the documentation](https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md) there are two commands to output the measured results of your bed mesh

> BED_MESH_OUTPUT: This command outputs the current probed z values and current mesh values to the terminal.
BED_MESH_MAP: This command probes the bed in a similar fashion to BED_MESH_CALIBRATE, however no mesh is generated. Instead, the probed z values are serialized to json and output to the terminal. This allows octoprint plugins to easily capture the data and generate maps approximating the bed's surface. Note that although no mesh is generated, any currently stored mesh will be cleared.

sennalb, 2019-09-01:
It seems to me to have the same problem on my delta printer, during the BED_MESH_CALIBRATE, offsets_y and offset_y of [probe] are not considered. It is correct ?

Arksine, 2019-09-01:
That is correct.  Given the confusion with this I am currently working on an update that will take the offsets into account when generating the probe coordinates.  The tricky part is that it is still possible to do a manual probe when you have an automatic probe configured, so I probably need to generate two sets of points for users that have a probe configured.

sennalb, 2019-09-03:
Ok, it seems a good solution.

KevinOConnor, 2020-01-11:
FYI, the latest bed mesh code now takes positions that could be thought of as "relative to the probe" (~commit edae40c3).

-Kevin

closed 2020-01-23 by KevinOConnor

bevanweiss, 2021-11-17:
Sorry to comment on a closed issue, but I'm new to klipper, and I'm confused by this.

I have a BLtouch that is to the left and forward of the nozzle (as I view the bed from the front).
Hence I have x_offset: +41.5, y_offset: +8.7, z_offset: +1.9
i.e. when probe is deployed the nozzle is X+41.5, Y+8.7, Z+1.9 from the probe tip

I then have a mesh_min of 40, 20, and stepper_x: position_min: 5
When I try to run the mesh bed levelling I get an error that the first position is unreachable, and it appears to be trying to position to -1.5, 11.3, 5
So it seems like the mesh bed levelling is trying to position the nozzle as though it was on the other side of the probe (i.e. as though the probe was on the right of the nozzle).

Is there perhaps an inconsistency in the direction of the _offset values?  i.e. x_offset and y_offset are probe with regards to the nozzle, and z_offset is nozzle with regard to the probe?

I suspect it would 'just work' if I put x_offset: -41.5, y_offset: -8.7.. but this seems 'wrong' when the z_offset is positive (i.e. nozzle is +Z above probe point).