Klipper3d/klipper #1865: Extrude below minimum temp after homing

J2xWo, 2019-08-06:
When starting a print, the bed heats up, when the desired temp is reached the target hotend temp is set and the printer starts to home. 
During homing (between the two hotend target temps in the graph) no temperature measurement is done (according to the graph) despite the bed and hotend heating. The desired bed temp was 60°C, after homing 62°C. So no temperature control has taken place. The desired hotend temp is 200°C. When homing is done it throws the "Extrude below minimum temp" error. Did i configure something wrong?

[klippy.log](https://github.com/KevinOConnor/klipper/files/3471929/klippy.log)
[image]

FHeilmann, 2019-08-06:
before doing any extrudes you should wait for your hotend to reach the target temperature. My guess is the following:
1. your gcode sets the target bed temperature and waits for it to be reached using `M190`
2. after the bed temperature is reached, your gcode sets the target extruder temperature using `M104`
3. The printer homes
4. After the homing procedure the printer tries to extrude.

The issue here is that your hotend does not reach the target temperature during the homing process (even though no temperature is reported during the homing process, the printer is still heating. So once the homing process finishes, the hotend is still too cold causing the error.

There are two ways to fix this: 
- **Replace** the `M104` g-code in your start script with a `M140`, this will cause the printer to wait for the hotend to heat up before starting the homing process
- **Add an additional** `M140` after your homing command (`G28`), this means that the printer will heat up while homing, and wait for the target temperature to be reached AFTER the homing process is done.

If this doesn't solve your problem please also share an example g-code file that produces the error.

Hope that helps,
Florian

J2xWo, 2019-08-07:
Yepp that did the trick! Thanks! Although your suggestion wasnt quite right. I had to replace `M104` with `M109`. Now it works tho

closed 2019-08-07 by J2xWo

FHeilmann, 2019-08-07:
I keep mixing those 4 commands up ^^ glad you figured it out. Happy printing!