Klipper3d/klipper #1868: SKR 1.3 with reprap discount GLCD just beeps/doesn't display.

Ph0rkoz, 2019-08-08:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3479461/klippy.log)

I used the sample SKR config and set the LCD pins as listed. When the printer starts with the klipper firmware stub it just beeps constantly and the screen lights up but nothing ever shows up. 

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/3479463/printer.cfg.txt)

It happens before klipper even connects and also after. I have the cables flipped of course as I came from a working config in marlin. 

Oh boy is there a learning curve. Z probe offset is positive (undocumented), not negative, hopefully the other offsets for X,Y are the same in regards to that. No software endstops even against Gcode, eg my Ymin is 5 and gcode of 0 tries to crash plate into the back.  But I get it, work in progress....

Also, that paper test is a PLA thing, I use the bed at 115c and it has to be heated up, thermal expansion is way more than 100 microns. I use an actual feeler gauge and with the probe script there is no way to add that increment to Z. Even better, it moves your nozzle over to where the probe was, with an offset of 46 to the right that's almost off the bed.

I must commend the maximum Z movement however, it really helps prevent crashes into the bed. Marlin will keep pushing until it breaks something.

Hywelmartin, 2019-08-08:
You haven't set set up any LCD.. so...
Undocumented z probe offset https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md#calibrating-probe-z-offset
Try a negative homing position...
The paper test is for the hotend thermal expansion(v6) and have never taken the bed in consideration..

Ph0rkoz, 2019-08-08:
I commented out the LCD because it wasn't working. I wish it were that simple. I tried the CR-10 one too just for kicks.

Yes, I read all of that. On marlin if the the nozzle is below the sensor the offset is negative. On klipper it is positive. The X is positive if the sensor is to the right, negative to the left. Y is negative if behind, positive if in front. I have no information on those last 2 in the documentation, hoping they are the same.

Ph0rkoz, 2019-08-08:
I got it to work, here is how:

Unplug right side LCD connector (the one near the encoder)
Enable RRD configs. 
Plug back in LCD: no beeeep, no display
Press encoder button, garble comes up.
Press reset on the LCD: Beeeeeeep
Press fw restart in octoprint.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3483547/klippy.log)

Ph0rkoz, 2019-08-11:
LCD has been working now for a day.

closed 2019-08-11 by Ph0rkoz