Klipper3d/klipper #1869: List of incompatible slicer features in Klipper.

JohnEdwa, 2019-08-08:
As Klipper works differently from many other firmwares, this creates some compatibility issues. To help users (and me) get better quality prints, I thought gathering them all and then creating a documentation/FAQ entry would make sense. As I only use Cura I can't say anything for other slicer, but maybe someone else will. 
Also, some of these are Cura defaults - it depends on the individual printer definition.

**Klipper motion planner deaccelerates completely when switching from extruding move to a travel move, and vice versa:**

*  Cura: Outer Wall Wipe Distance [(Image 1)](https://user-images.githubusercontent.com/5417183/62705391-ff212880-b9f5-11e9-866d-1a6634af1387.png): 
Generates a travel move from the end of the outer wall over the start of it and only then moves inside the wall, in an effort to blend them together and reduze the Z-sea,. Klipper will deaccelerate and stop to the end of the wall and then accelerate over the start. This causes a small ooze blob.
* Cura: Coasting [(Image 2)](https://user-images.githubusercontent.com/5417183/62705505-3263b780-b9f6-11e9-9826-c35969f5c082.PNG): 
Stops extrusion slightly before the wall ends to reduce Z-seam blob by letting ooze do the last bit. Causes Klipper to deaccelerate to a stop at this point, actually creating a larger Z-seam. If combined with Outer Wall Wipe Distance above, actually creates *two* blobs.
* Cura (minor slowdown) : Top/Bottom Pattern, Lines [(Image 3)](https://user-images.githubusercontent.com/5417183/62705612-63dc8300-b9f6-11e9-9166-abef029b0956.PNG): 
Klipper comes to a full stop before and after each line.

**Pressure advance activates fully when line width changes (#1758):**

* Cura: Fill Gaps Between Walls [(Image 4)](https://user-images.githubusercontent.com/5417183/62705810-dcdbda80-b9f6-11e9-959c-813097cc1dd5.PNG): 
Generates multiple short extrusions of different widths to fill gaps, each of which will also result in a full pressure retraction. NOTE! Disabling this will result in weaker prints as any area smaller than your line width will simply be left empty. [(Image 5)](https://user-images.githubusercontent.com/5417183/62705933-22000c80-b9f7-11e9-950f-68fd6a24f9e7.PNG).
* Cura: Compensate Wall Overlaps and Minimum Wall Flow [(Image 6)](https://user-images.githubusercontent.com/5417183/62706019-4b209d00-b9f7-11e9-8a40-83db384f5bd9.PNG) 
**Do not use this feature with pressure advance.**
Compensates two walls overlapping by reducing line with. The same issue as Fill Gaps Between Walls, but now on **all** walls of your print.
* Cura (minor annoyance) : Top/Bottom Pattern: Lines: 
Pressure advance retraction after each line. (which is technically working as intended).

**G-code incompatibilities:**

* Cura: G29: Any printer that has ABL will use the standard G29 in their start script for it.
* Cura: M600: Post Processing script  "Filament Change" uses M600, which is not implemented.
* Cura: M201, M203, M204, M205: The new Creality definitions set these in the start Gcode, Klipper only supports ``M204 S<value>``:
``M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration``
``M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate``
``M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration``
``M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk``
* Cura (partial): M84: The end Gcode command ``M84 X Y E`` tries to leave Z energized, but Klipper turns them all off.

---

Here bot, take this fake log. [klippy.log](https://github.com/KevinOConnor/klipper/files/3482042/klippy.log)

Ph0rkoz, 2019-08-08:
Simplify3d will extrude more than max_extrude_cross_section, it has come up in issues before. I set it to 96, hopefully it handles everything throughout the code. Wouldn't want a multi hour print stopped because of this and so I'm hesitant to try anything large yet.

The lack of soft end stops applied to Gcode also gives me pause. I've hit my sensor on the Z axis bars because its so far to the right and had to reduce my bed area. No idea if its taken into account when motion planning.

danielfmo, 2019-08-09:
@JohnEdwa nice work pointing out problematic settings in Cura.
Thought from your text is it not clear to my which setting to use instead. 
For example, coasting I can deduce from the text that it is best to disable but not clear. And for Outer Wall Wipe Distance it is not clear which value is it recommended to use.

Thank you.

JohnEdwa, 2019-08-09:
Outer Wall Wipe, Coasting and Compensate Wall Overlaps should all be disabled, as they are meant to improve visual quality but having them on with Klipper will create even larger issues. It won't look as nice as if they worked properly and it does mean you'd get a smaller and less visible Z-seam if you used some other firmware than Klipper.

Fill Gaps Between Walls is a little more complicated, because disabling it creates much weaker prints and can leave visible gaps ([like the deck-box on this benchy](https://www.3dhubs.com/talk/uploads/db5060/original/2X/c/cc9cc4183f9e1dab8a78165ff050c5eb60e4b927.JPG)), but having it on abuses the extruder motor as it will be instructed to repeatedly do tens of pressure releases within a few millimetres of a wall, which certainly won't be good for it long term. And on a 3 to 1 extruder like a BMG, it's also really loud.   
So the print will look and be better with it on, but on the flip side you are wrecking your printer.

nophead, 2019-08-09:
It might wear a geared extruder but stepper motors aren't wrecked by rapid movements. They either stall or work if the acceleration and jerk are set appropriately.

KevinOConnor, 2019-08-10:
Thanks.  FYI, there is some info on configuring slicers at https://www.klipper3d.org/Slicers.html .  I agree it would be good to expand it with more specific information.

-Kevin

tomohulk, 2019-08-11:
@JohnEdwa do you just set the outer wall wipe distance to 0?  Or is there a way to actually disable it?

drphil3d, 2019-08-12:
I did a git pull last night before printing this part and too be honest I've never had it run so horribly. https://photos.app.goo.gl/RmXTXP87hZXhKYZA6

dstulken, 2019-08-13:
@JohnEdwa, No offense, but are you sure that Cura's features aren't a red herring for a printer that needs more calibration?

Think about this - How is the behavior with outer wall wipe distance enabled any different from any other extrusion end?  Even if outer wall wipe distance was zero, the end of the outer wall STILL has to stop extruding and change to a travel move at that same point, to get to its next extrusion location.  

So if zeroing out the outer wall wipe distance doesn't eliminate the motion stop, and Klipper fully decelerates anyway regardless of whether outer wall wipe distance is non-zero, then any *extra* blobbing you see in that spot with outer wall wipe enabled must be material oozing during the following travel move.   

I would suspect that enabling outer wall wipe is not a Klipper incompatibility per-se, but instead is highlighting an ooze problem on your machine- perhaps temps, retraction, or pressure advance could be tweaked to improve things?

(Keeping in mind that a *small* amount of oozing *is* the desired effect for outer-wall-wipe, as that is what fills the start/stop gap.   Also note that I'm not suggesting that Klipper's handling of this is ideal, as transitioning without the full motion stop would likely produce a smoother outer surface finish...)

JohnEdwa, 2019-08-13:
@dstulken 

When I say "small blob" I mean *a small* blob. This originates from the issue that was demonstrated by the "Spiralize" feature in Cura when it had a bug that made it generate sub-millimeter travel moves and each one made a tiny blob that was visible on the otherwise perfectly smooth surface, because Klipper deaccelerated, stopped and accelerated for each of them. It wasn't so much an actual ooze blob, but just that it made the wall thicker at that point.

Marlin has a smooth and fast extrude - move -  extrude when it goes from the inner to the outer wall, which usually results in the start of the wall being slightly under extruded, so having the wipe at the end fills that gap perfectly and makes the seam smaller.

Klipper instead has to completely stop at the beginning of the wall which creates a larger start, and it stops at the end which creates a larger end. Without the wipe it then immediately starts going back inside the wall, but with the wipe enabled the acceleration will happen on top of the already printer wall, making it slightly wider again.    
Having Coasting enabled means it starts this process even sooner, so the end blob will be further away from the start making the seam look much wider and more visible.

They aren't huge issues like using "Compensate Wall Overlaps" with Pressure Advance because of #1758 , but they don't work properly with Klipper either so they shouldn't be used.

dstulken, 2019-08-13:
Thanks @JohnEdwa.
I see what you're saying, and appreciate the explanation, but I'm still not sure that outer wall wipe is an incompatibility, or something that should be suggested be turned off as general advice.   Whether it makes things better or worse depends on how you value your surface finish - blobs vs gaps - and how strongly you see either of those artifacts (or neither at all) depends highly on your printer, materials, speeds, temperatures, etc.  

#1758 is indeed a significant issue - disabling gap fill would be a huge problem for many things I print - so I do hope that that is able to be resolved.  I would consider #1758 to be a genuine Klipper issue, not a slicer incompatibility as a result of differing design philosopies, and again would not advise disabling it in the slicer unless absolutely necessary...  (For example, I'd rather have my extruder hammer away furiously than have a structurally deficient print.  Neither is a *good* choice.)

In all of these cases, until the issues can be resolved, regardless of where fault lies, I fully agree with you that documenting the effects is helpful.   They just need to be phrased such that the users understand that it isn't necessarily their slicer that is broken, that those features aren't bad in concept or purpose, and that the advice documented reflects the expected outcome with the current state of Klipper's motion/extrusion planning.

JohnEdwa, 2019-08-13:
> They just need to be phrased such that the users understand that it isn't necessarily their slicer that is broken, that those features aren't bad in concept or purpose, and that the advice documented reflects the expected outcome with the current state of Klipper's motion/extrusion planning.

You obviously can't blame the slicers as these issues are from Klipper doing things differently from every other firmware out there. I understand why Klipper does what it does, but preferably there would be a way to make Klipper work like the others so it would be compatible with mosre stuff.

And you make a good point that would need more testing - even though a feature doesn't work the same way, it doesn't mean it doesn't work at all. Though for the other features, I think I'm mostly correct.

DanielJoyce, 2019-08-22:
looks like the slicer settings page for Klipper already mentions this behavior:
```
Do not enable “coasting”
The “coasting” feature is likely to result in poor quality prints with Klipper. Consider using Klipper’s pressure advance instead.

Specifically, if the slicer dramatically changes the extrusion rate between moves then Klipper will perform deceleration and acceleration between moves. This is likely to make blobbing worse, not better.

In contrast, it is okay (and often helpful) to use a slicer’s “retract” setting, “wipe” setting, and/or “wipe on retract” setting.
```

oullah, 2019-09-08:
@JohnEdwa this post seems to be contradictory what the Klipper [slicers](https://github.com/KevinOConnor/klipper/blob/master/docs/Slicers.md) page says about outer wall wipe: 

> In contrast, it is okay (and often helpful) to use a slicer's "retract" setting, "wipe" setting, and/or "wipe on retract" setting

JohnEdwa, 2019-09-08:
Using `Outer Wall Wipe` with Klipper "works", in the sense that it stops stringing between parts as it is designed to. Without it when the outer wall ends, if the next part is outside the wall and your retraction settings aren't perfect it'll drag a tiny whisp between the two parts.

But because it does the deacceleration/stop/acceleration/wipe thing, it leaves a small blob just before the seam making it larger, and it does it even on parts that don't benefit from the wipe at all.

oullah, 2019-09-08:
Oh, hmm, I guess I misunderstood what `Outer Wall Wipe` was for. I thought it was to help prevent blobs at z seams.

JohnEdwa, 2019-09-08:
In Marlin and other firmwares that don't do the stop as Klipper does, it helps minimize the seam, as it uses the "oozing" of the nozzle to fill the gap left by the start of the outer wall. But Klipper isn't compatible with it due to how the motion planner it uses works differently from everything else and the resulting seam.   
Klipper will stop before and after the outer wall line as it transitions from a move to extrusion and back to a move, and both stops will create a slightly larger zit, so there shouldn't be under extrusion to fill and stopping sooner as the wipe makes it do will make the end zit even larger.

KevinOConnor, 2019-09-21:
FWIW, PR #1997 should help address many of the concerns raised here.  That new code is in a preliminary state, but is available for those wishing to run tests.

-Kevin

KevinOConnor, 2019-11-21:
With the merge of #1997 I believe most/all of the items raised here no longer apply.

-Kevin

JohnEdwa, 2019-11-21:
Yup, fairly sure it fixes all the motion planning issues.
Only ones left are the Gcode ones, and as discussed elsewhere they can be implemented as macros.

With the exception of M84 I think?

Does Klipper currently have a way to turn off individual steppers or is it all at once?    
I know that it's something that's needed with some printers where disabling Z will cause the gantry to crash down, and recently I stumbled on [this dual-extruder kit](https://www.3dchameleon.com/) that uses a mechanical switch to swap extruder motors, something I think shouldn't be done if the extruder stepper is energized.

KevinOConnor, 2019-11-21:
> Does Klipper currently have a way to turn off individual steppers or is it all at once?

Not currently.  Raised a few times (eg, #1947).  If I get time I'll add an option to FORCE_MOVE to allow a user to explicitly enable/disable a stepper.  If someone else gets to it first then that's great.

-Kevin

JohnEdwa, 2019-11-21:
Adding them as compatible parameters to M18 and M84 would be very useful as they are already used in many default end G-code scripts - as an example, Cura has "M84 X Y E" for every single Creality machine so they keep Z energized after a print.

JohnEdwa, 2019-11-22:
I'll close this as the incompatibilities are either gone or already under discussion/inactive under other issues.

closed 2019-11-22 by JohnEdwa

WarriorClub, 2020-09-12:
I know this is a closed thread but I thought I would mention that some incompatibilities mentioned seem compatible based on a number of test prints. Disabling "compensate wall overlaps" feature in cura caused much headache as I was trying to figure out why walls were looking underextruded (indented lines) especially on layers with additional corners inside like part of a letter on the calibration cube. Also I noticed alot of messed up corners. Enabling this drastically improved these areas and eliminated this altogether on some. This feature i believe should not be listed as an incompatibility.