Klipper3d/klipper #1870: Probe triggered prior to movement

Saso-lab, 2019-08-09:
I'm tring to run the delta_calibrate, but the nozzle stops few centimeters above the bed and it reports the error "probe triggered prior to movement". I tried to insert a pause with the following macro, without success. If i run the second time the delta_calibrate command it starts probing.
I don't know if this can influence the calibration process, but since i had this issues i'm not able to have a perfect fist layer. These are my result after a mesh calibrate but when i print large objects i can see some points on the bed where the nozzle is too close (even if the bed map does not seem bad).
I'm using an skr 1.3 with smart effector.
[image]
[image]
[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/3485286/klippy.log)

Hywelmartin, 2019-08-09:
have you tried with the pause inside the [probe] section??
```
[probe]
pin = P2.0
x_offset = 0
y_offset = 0
z_offset = 0
activate_gcode:
	G4 P500
```

Saso-lab, 2019-08-09:
I just tried without success, i'm having this issue since i switched to the skr 1.3

Hywelmartin, 2019-08-09:
pullup? as you are using the servo pin
pin: ^!2.0

Saso-lab, 2019-08-10:
Thanks a lot man, i missed the "^". 
Now everything works.

closed 2019-08-10 by Saso-lab