Klipper3d/klipper #1875: Change attribute criteria for min_speed from TemperatureFan

meteyou, 2019-08-10:
Can you change the line 27 in extras/temperature_fan.py?

From:
> self.min_speed = config.getfloat('min_speed', 0.3, above=0., maxval=1.)

to:
> self.min_speed = config.getfloat('min_speed', 0.3, maxval=1.)

[klippy.log](https://github.com/KevinOConnor/klipper/files/3489241/klippy.log)

KevinOConnor, 2019-08-17:
It's not clear to me what the underlying issue being reported here is.  The "above=0." is intended to catch obvious config errors (like a user configuring `min_speed=-1234`).

-Kevin

meteyou, 2019-08-17:
with above=0. your min speed is 0.1 but i want to setup the fan with min speed 0.0. This is my problem with above=0.

KevinOConnor, 2019-08-23:
I believe that code was written by @wizhippo - any comments on this?

-Kevin

KevinOConnor, 2019-09-14:
FYI, I made the requested change (commit 43103929).

-Kevin

closed 2019-09-14 by KevinOConnor