Klipper3d/klipper #1876: BL-Touch triggering too soon sometimes

NicoSoftware, 2019-08-11:
Hey,
I'm very happy with using klipper and it's working pretty reliable.
I only have a problem with my TL BL-Touch Clone.
when I bed_mesh_calibrate I'll get the right z-distance about 70% of the time.
Sometimes my probe just triggers to soon without touching the bed surface.
a bed probe looks like this most of the time:
'''
Recv: // probe at 50.000,50.000 is z=0.615000
Recv: // probe at 50.000,50.000 is z=0.610000
Recv: // probe at 50.000,50.000 is z=0.607500
Recv: // probe at 111.662,50.000 is z=0.525000
Recv: // probe at 111.662,50.000 is z=0.535000
Recv: // probe at 111.662,50.000 is z=0.537500
Recv: // probe at 173.325,50.000 is z=0.467500
Recv: // probe at 173.325,50.000 is z=0.495000
Recv: // probe at 173.325,50.000 is z=0.495000
Recv: // probe at 234.975,50.000 is z=4.237500
Recv: // probe at 234.975,50.000 is z=0.502500
Recv: // probe at 234.975,50.000 is z=0.520000
Recv: // probe at 234.975,108.325 is z=0.475000
Recv: // probe at 234.975,108.325 is z=7.717500
Recv: // probe at 234.975,108.325 is z=0.442500
Recv: // probe at 173.325,108.325 is z=0.502500
Recv: // probe at 173.325,108.325 is z=0.507500
Recv: // probe at 173.325,108.325 is z=0.505000
Recv: // probe at 111.662,108.325 is z=0.582500
Recv: // probe at 111.662,108.325 is z=0.580000
Recv: // probe at 111.662,108.325 is z=0.582500
Recv: // probe at 50.000,108.325 is z=0.655000
Recv: // probe at 50.000,108.325 is z=0.647500
Recv: // probe at 50.000,108.325 is z=0.645000
Recv: // probe at 50.000,166.662 is z=0.705000
Recv: // probe at 50.000,166.662 is z=0.707500
Recv: // probe at 50.000,166.662 is z=0.702500
Recv: // probe at 111.662,166.662 is z=0.595000
Recv: // probe at 111.662,166.662 is z=0.600000
Recv: // probe at 111.662,166.662 is z=7.845000
Recv: // probe at 173.325,166.662 is z=0.492500
Recv: // probe at 173.325,166.662 is z=0.507500
Recv: // probe at 173.325,166.662 is z=0.507500
Recv: // probe at 234.975,166.662 is z=0.445000
Recv: // probe at 234.975,166.662 is z=0.450000
Recv: // probe at 234.975,166.662 is z=0.445000
Recv: // probe at 234.975,224.987 is z=0.485000
Recv: // probe at 234.975,224.987 is z=0.487500
Recv: // probe at 234.975,224.987 is z=0.485000
Recv: // probe at 173.325,224.987 is z=0.575000
Recv: // probe at 173.325,224.987 is z=0.570000
Recv: // probe at 173.325,224.987 is z=0.567500
Recv: // probe at 111.662,224.987 is z=0.670000
Recv: // probe at 111.662,224.987 is z=7.910000
Recv: // probe at 111.662,224.987 is z=0.650000
Recv: // probe at 50.000,224.987 is z=0.765000
Recv: // probe at 50.000,224.987 is z=0.767500
'''
I already played around with those bltouch section properties:
'''
#pin_up_reports_not_triggered: True
#   Set if the BLTouch consistently reports the probe in a "not
#   triggered" state after a successful "pin_up" command. This should
#   be True for a genuine BLTouch; some BLTouch clones may require
#   False. The default is True.
#pin_up_touch_mode_reports_triggered: True
#   Set if the BLTouch consistently reports a "triggered" state after
#   the commands "pin_up" followed by "touch_mode". This should be
#   True for a genuine BLTouch v2 and earlier; the BLTouch v3 and some
#   BLTouch clones require False. The default is True.
'''
Does someone had similar issues and a solution to this?
Are there some other properties in the config I can play around with?
Thanks a lot in advantage

KevinOConnor, 2019-08-17:
There's been a lot of reports of inconsistent measurements with BLTouch clones.  It is believed to be poor manufacturing of the clones.

On the software side, you could try to detect and filter out bad values with something like:
```
[bltouch]
...
samples: 3
samples_tolerance: 0.005
samples_tolerance_retries: 3
```

The details are in the (recently updated) config/example-extras.cfg file.

-Kevin

NicoSoftware, 2019-08-20:
```
[bltouch]
...
samples: 3
samples_tolerance: 0.005
samples_tolerance_retries: 3
```
Thanks alot, that's exactly what I need!
```
The details are in the (recently updated) config/example-extras.cfg file.
```
Thanks for taking the time, I highly appreciate it.
Regards
Nico F.