Klipper3d/klipper #1877: Extruder Problems after Update, Yesterday

Achim01, 2019-08-11:
Objektsize 60x75x0.2 for Test.
(first layer is the Problem)

Printing with different amperage for Extruder
brings the same Result.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3489857/klippy.log)
[image]

[60x75x02.gcode..zip](https://github.com/KevinOConnor/klipper/files/3489859/60x75x02.gcode.zip)

KevinOConnor, 2019-08-17:
Nothing in the Klipper extruder code has changed recently.  You may wish to try running `DUMP_TMC STEPPER=extruder` to see if there is anything indicating an issue with the extruder driver.

-Kevin

KevinOConnor, 2019-09-06:
I'm going to close this issue as it appears the conversation has concluded.

-Kevin

closed 2019-09-06 by KevinOConnor