Klipper3d/klipper #1878: Parts fan dropping temperature only while printing

CC192, 2019-08-11:
Hi there. I am quite new to klipper an ran into an issue I couldn't solve .
First of I did the calibrations as described including several PID tunings. Somehow though I can only print without parts cooling fan on because the temperature will drop enormously. With Marlin I never had that issue and I even PID tuned the settings in with parts fan running on full speed even though I use it at 70% at max while printing.

So now here comes the strange part: If I heat up the printer via manual input in Octoprint and then turn on the fan, things will work as used in Marlin. I also thought about the movement or filament push-out causing the temp loss, but I tried my best simulating the print over the terminal (e.g. extruding while G0-moving the printhead)

Is there something I am missing out here?

[klippy.log](https://github.com/KevinOConnor/klipper/files/3490502/klippy.log)

KevinOConnor, 2019-08-17:
This is generally indicative of the power supply not being able to provide enough power to the heaters.  Consider adding something like `max_power=0.250` to the [heater_bed] section of the config to reduce the overall amount of power used on the bed heater.  You may also wish to configure a max_power setting in the fan (if it's providing too much cooling).

-Kevin

CC192, 2019-08-18:
Thabk you. Is Klipper in need of more Power to heat than Marlin? Because I did print PETG with those settings a few days prior to that. My PSU even has moee Power than the stock one (ATX over PiHat)

CC192, 2019-08-19:
I replaced the Heat Cartridge with a 40W version. It works on Marlin now. Even though I can't start Klipper anymore because there iis no tmp/printer, this issue can be closed, because it seems to be a hardware malfunction.

CC192, 2019-08-28:
It has been. Second one as well after clean reinstall