Klipper3d/klipper #1879: Can't move after canceled/failed print

Ph0rkoz, 2019-08-11:
So I'm doing the pressure advance test.  Doing mulitiple prints in a row up to a certain height and then canceling. After that I must home the printer to move again.

In the case of a canceled or failed print with the object blocked, it would sure be nice to move Z up and away. In more fun cases, you will end up with nozzle burried in the print, unable to move.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3490517/klippy.log)

mental405, 2019-08-12:
Cancelling a print is a function of OctoPrint and not Klipper. 

There is a "GCODE Scripts" section in the settings for OctoPrint that allows you to specify what happens after a print job is cancelled. I believe by default it issues an M84 command. In Klipper, if the motors are disabled, you will have to rehome the axis or axes before you can command the steppers to move again.

In the case of your moving the Z up and away add something to the aforementioned script section like

G91
G1 Z30 F3000

to move the nozzle away from the print after it is cancelled. Alternatively a gcode macro could be written in the Klipper config to serve the same function, but you will still need to call the macro from the cancel print action.

Ph0rkoz, 2019-08-12:
Ok, that makes sense, maybe I should just have it not turn off the motors and move the Z up by a few like you did.

dstulken, 2019-08-22:
If you are repeatedly stopping the print early on purpose, maybe changing your slicer settings would be a more appropriate solution than modifying the OctoPrint cancel behavior?

For example, if you always stop the print at 50% height, you could replicate the exact same end result by instead scaling the model Z by 50% and setting the top layer count to zero in your slicer.

closed 2019-08-26 by Ph0rkoz