Klipper3d/klipper #1880: Can not update MCU 'mcu' config as it is shutdown

JeepersMD, 2019-08-12:
Hello,

Please understand this has been an extreme learning process but I have fought it all the way with new board install and such.  Running Bigtreetech SKR 1.3 with 2130 for x,y,z on an anycubic kossel delta pulley.  I finally got it to read all the way thru and hit this error brickwall that I can't figure out.  I just can't get past it.  I am attaching klippy log.  I look forward to any help at all.  Thanks. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3490595/klippy.log)

BlackStump, 2019-08-12:
do you have a thermistor connected to the heated bed?
because the error message in your log is 

MCU 'mcu' shutdown: ADC out of range

and your bed temp is -99.6

JeepersMD, 2019-08-12:
I do. Could it be reversed or ? As I said very new to this.

tgmorris99, 2019-08-12:
The plug for the thermistor has no real polarity. Do you have it plugged in to the correct spot?

JeepersMD, 2019-08-12:
Ok, you were absolutely right.  The bed wires had come loose.  So I was able to get thru and start the Klipper config checks.  I am having one other issue.  If I need to start in another topic I can, but attaching new klippy log.  I can't get steppers to do anything.  Even with the terminal commands in the checks guide.  I am running TMC 2130's on x, y, z on a delta so they have been labeled a, b, c in both upper config and stepper.  Let me know what you think.  And thanks again for help this far. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3494409/klippy.log)

JeepersMD, 2019-08-12:
Actually the error is endstop a still triggered after retract and must home first ??

KevinOConnor, 2019-08-17:
You should run through the checks described at https://www.klipper3d.org/Config_checks.html and follow the instructions there on any failed check.

-Kevin

KevinOConnor, 2019-09-06:
I'm going to close this issue as it appears the conversation has concluded.

-Kevin

closed 2019-09-06 by KevinOConnor