Klipper3d/klipper #1883: Safe Z Homing with BLTouch

quintinjvr, 2019-08-12:
Good day

I've been trying to find a safe way to home the Z-Axis without using separate GCODE commands. I am using a BLTouch and after the X and Y homing, the BLTouch is not above the bed and can cause quite alot of damage when doing a Z-Home command while not in position.

Could anyone please give some recommendations as to how I can implement a safe check for X and Y to move to the correct position before allowing Z axis homing? Similar to the Safe Z Homing function (Which I couldn't get to work though).

I previously made use of the TMC2130 sensorless homing. The retract distance used to stop in the correct position, but after having issues with the sensorless homing being inaccurate I switched over to normal limit switch, but now I face the above issue with Z- Homing.

Thank you in advance.
Q

[klippy.log](https://github.com/KevinOConnor/klipper/files/3494240/klippy.log)

BlackStump, 2019-08-12:
https://github.com/KevinOConnor/klipper/blob/master/config/sample-probe-as-z-endstop.cfg#L24

quintinjvr, 2019-08-13:
Hi @BlackStump 
Thank you for the code extract, I've tried that and it is working. 

I'm still curious to know if the [Safe_Z_Homing] function is possible with the BLTouch?

https://github.com/KevinOConnor/klipper/blob/3d38211b8ab0b4736cdcd7e4b3f3a05bb598db3e/klippy/extras/safe_z_home.py

FHeilmann, 2019-08-14:
I don't see why the bltouch shouldn't work with safe z homing. Have a look in the example_extras.cfg file to see how to set up a specific xy homing coordinate. Everything else should remain untouched. Make sure to remove any homing override sections you may have in your config.

quintinjvr, 2019-08-28:
Hi Kevin

Yes the issue was solved. 
The thread can be closed.

Thanks for the help 

Q

closed 2019-08-28 by quintinjvr