Klipper3d/klipper #1884: On SKR mini v 1.1  and latest klipper - issues device descriptor read/64, error -32

vladbabii, 2019-08-13:
compiled with latest version - effe6f6ddd5132db4ab4f88bf8616aab575e7691

Dmesg output when connecting to usb
```
[  521.874046] usb 1-1.2: new full-speed USB device number 6 using dwc_otg
[  521.974049] usb 1-1.2: device descriptor read/64, error -32
[  522.194049] usb 1-1.2: device descriptor read/64, error -32
[  522.414056] usb 1-1.2: new full-speed USB device number 7 using dwc_otg
[  522.514053] usb 1-1.2: device descriptor read/64, error -32
[  522.744162] usb 1-1.2: device descriptor read/64, error -32
[  522.864138] usb 1-1-port2: attempt power cycle
[  523.524074] usb 1-1.2: new full-speed USB device number 8 using dwc_otg
[  523.964085] usb 1-1.2: device not accepting address 8, error -32
[  524.294097] usb 1-1.5: reset high-speed USB device number 5 using dwc_otg
[  525.174460] usb 1-1.4: USB disconnect, device number 4
[  525.175135] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[  525.175216] ch341 1-1.4:1.0: device disconnected
```

With this version everything works fine
```
commit c176b66f291371e8ba1c171e56bb2a0416ccf16f
Author: Kevin O'Connor <kevin@koconnor.net>
Date:   Sun Jul 28 22:27:32 2019 -0400

    stm32f4: Allow the USART serial port to be configured

    Signed-off-by: Kevin O'Connor <kevin@koconnor.net>
```

I will try to compile with newer  versions than above and see when things broke, and will post here with results. As far as can tell it should be about when make menuconfig added options for different cpu types for STM32F103

Please let me know what information i can provide, since klipper logs is useless because the service is not even started.

Thank you

vladbabii, 2019-08-13:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3495740/klippy.log)
Fake klipper log for bot

KevinOConnor, 2019-08-16:
There's been a bunch of changes to the stm32 code recently.  However, the USB support on the stm32f103 was not changed.  The config options have changed a bit, though, so make sure you have a clean checkout, be sure to rerun "make menuconfig", and make sure the appropriate config settings are in place (basically, select stm32f103, 28KiB bootloader, and otherwise use the defaults).

If you continue to have issues, can you attach your .config file here?

-Kevin

vladbabii, 2019-08-19:
I believe the printer config file doesn't really matter because the USB serial connection is not detected on the OS level for the klipper software be able to use it. I have attached it here. 
[printer.cfg.zip](https://github.com/KevinOConnor/klipper/files/3515244/printer.cfg.zip)

The printer is `/dev/printerhill` because i have a file in `/etc/udev/rules.d/99-printerhill.rules` with:
`SUBSYSTEM=="tty", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="abcd", SYMLINK+="printerhill"`

It's a testing platform for the X/Y part of Corexy and it moves and works corectly with the firmware version mentioned above.

I will re-compile from latest in about 8 hours hours. If its any help I can provide remote ssh access to a pi or laptop connected to that board and can help you debug.

I have used the 28kb bootloader and noticed the change in the menuconfig options - that prompted me to try older klipper versions because when i bought the mini boards (about one and a half months ago) i managed to get it running with the older menuconfig options.

vladbabii, 2019-08-19:
```
git clone https://github.com/KevinOConnor/klipper .
make clean
git describe --tags
> v0.7.0-769-ge817b85
make menuconfig
```

then selected microcontroller "stmicroelectronics stm32f1/f4"

then selected processor model stm32f103

then selected booloader offset 28kib

left default 'use usb for communication (instead of serial)'

usb ids:
- vendor id: 2341
- usb device id: 0xabcd
- usb serial number: 12345

```
make
```

got this error now (this is new)

```
ERROR: The compiler failed to correctly compile Klipper
It will be necessary to upgrade the compiler
See: https://bugs.launchpad.net/ubuntu/+source/newlib/+bug/1767223
```

```
apt update
apt upgrade -y
...
make clean
make menuconfig
```
made sure all settings remained then ran make
```
  Build Kconfig config file
  Creating symbolic link out/board
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/spi_software.o
  Compiling out/src/neopixel.o
  Compiling out/src/stm32/main.o
  Compiling out/src/stm32/watchdog.o
  Compiling out/src/stm32/gpio.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_timer.o
  Compiling out/src/../lib/stm32f1/system_stm32f1xx.o
  Compiling out/src/stm32/stm32f1.o
  Compiling out/src/stm32/adc.o
  Compiling out/src/stm32/spi.o
  Compiling out/src/stm32/usbfs.o
  Compiling out/src/generic/usb_cdc.o
  Building out/compile_time_request.o
Version: v0.7.0-769-ge817b85-20190819_205041-octopus
  Assembling out/src/../lib/stm32f1/gcc/startup_stm32f103xb.o
  Preprocessing out/stm32.ld
  Linking out/klipper.elf
  Creating hex file out/klipper.bin
```

```
git rev-parse HEAD
> e817b85b78b1482c1269ef6549e430b918da01b5
```

put file on microsd card and powered on board with 12v power and no usb

then shutdown 12v power, removed microsd and check if a file BIGTREETECH.bigtreetech is there - and it was

Now ran 
```
dmesg -c 
```
to clean dmesg output from linux. Then plugged usb wire back into the board

```
[554916.514653] usb 3-3: new full-speed USB device number 26 using ohci-pci
[554916.674834] usb 3-3: device descriptor read/64, error -32
[554916.942838] usb 3-3: device descriptor read/64, error -32
[554917.234828] usb 3-3: new full-speed USB device number 27 using ohci-pci
[554917.394970] usb 3-3: device descriptor read/64, error -32
[554917.662645] usb 3-3: device descriptor read/64, error -32
[554917.770740] usb usb3-port3: attempt power cycle
[554918.270859] usb 3-3: new full-speed USB device number 28 using ohci-pci
[554918.686864] usb 3-3: device not accepting address 28, error -32
[554918.870873] usb 3-3: new full-speed USB device number 29 using ohci-pci
[554919.286867] usb 3-3: device not accepting address 29, error -32
[554919.286993] usb usb3-port3: unable to enumerate USB device
```

next, ran dmesg -c and powercycled the board (12v off, 10 second wait, 12v on

dmesg output:
```
[554999.596155] usb 3-3: new full-speed USB device number 30 using ohci-pci
[554999.756177] usb 3-3: device descriptor read/64, error -32
[555000.024176] usb 3-3: device descriptor read/64, error -32
[555000.320192] usb 3-3: new full-speed USB device number 31 using ohci-pci
[555000.480054] usb 3-3: device descriptor read/64, error -32
[555000.748170] usb 3-3: device descriptor read/64, error -32
[555000.856261] usb usb3-port3: attempt power cycle
[555001.356208] usb 3-3: new full-speed USB device number 32 using ohci-pci
[555001.772197] usb 3-3: device not accepting address 32, error -32
[555001.956199] usb 3-3: new full-speed USB device number 33 using ohci-pci
[555002.372199] usb 3-3: device not accepting address 33, error -32
[555002.372339] usb usb3-port3: unable to enumerate USB device
```

vladbabii, 2019-08-19:
Next, reverted to old working firmware via sd card upload and repowered the board:

Dmesg output:

```
[555172.060626] usb 3-3: USB disconnect, device number 38
[555172.506799] usb 3-3: new full-speed USB device number 39 using ohci-pci
[555172.709877] usb 3-3: New USB device found, idVendor=2341, idProduct=abcd
[555172.709888] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[555172.709894] usb 3-3: Product: Klipper firmware
[555172.709901] usb 3-3: Manufacturer: Klipper
[555172.709907] usb 3-3: SerialNumber: 12345
[555172.713987] cdc_acm 3-3:1.0: ttyACM0: USB ACM device
```

This is on the same machine with everything the same (cable, usb port, board, etc).

vladbabii, 2019-08-19:
@KevinOConnor please let me know what else to try. 

Do you know on which revision the build menu for this processor changed? I'd like to make a build before and one after to see if that is the cause or something else.

Thank you for your time!

KevinOConnor, 2019-08-19:
> Do you know on which revision the build menu for this processor changed?

In ec3d865b it was possible to build both the old and the new stm32f103 code (the new code was under the stm32f4 menu option).  In the following commit (8b9cc623) the old build method was removed.

-Kevin

KevinOConnor, 2019-08-19:
Can you attach the ".config" file from the failed build?  (It is a file produced by "make menuconfig" - it should be in ~/klipper/.config .)

-Kevin

vladbabii, 2019-08-19:
[dot.config.zip](https://github.com/KevinOConnor/klipper/files/3517639/dot.config.zip)
Here you go.

vladbabii, 2019-08-19:
And this is from the working one - c176b66f291371e8ba1c171e56bb2a0416ccf16f
[working.dot.config.zip](https://github.com/KevinOConnor/klipper/files/3517644/working.dot.config.zip)

KevinOConnor, 2019-08-20:
Can you retry with the latest code (commit c930fc39)?

-Kevin

vladbabii, 2019-08-20:
In 5 minutes

vladbabii, 2019-08-20:
```
git clone https://github.com/KevinOConnor/klipper.git
git rev-parse HEAD
> c930fc392b6977b17d6f7953bd738583974208c2
make clean
make menuconfig
```
stm31f1/f2 -> smt32f103 -> 28kib 

.config contents:
```
#
# Automatically generated file; DO NOT EDIT.
# Klipper Firmware Configuration
#
# CONFIG_LOW_LEVEL_OPTIONS is not set
# CONFIG_MACH_AVR is not set
# CONFIG_MACH_ATSAM is not set
# CONFIG_MACH_ATSAMD is not set
# CONFIG_MACH_LPC176X is not set
CONFIG_MACH_STM32=y
# CONFIG_MACH_STM32F0 is not set
# CONFIG_MACH_PRU is not set
# CONFIG_MACH_LINUX is not set
# CONFIG_MACH_SIMU is not set
CONFIG_STEP_DELAY=2
CONFIG_BOARD_DIRECTORY="stm32"
# CONFIG_MACH_atmega2560 is not set
# CONFIG_MACH_atmega1280 is not set
# CONFIG_MACH_at90usb1286 is not set
# CONFIG_MACH_at90usb646 is not set
# CONFIG_MACH_atmega32u4 is not set
# CONFIG_MACH_atmega1284p is not set
# CONFIG_MACH_atmega644p is not set
# CONFIG_MACH_atmega328p is not set
# CONFIG_MACH_atmega328 is not set
# CONFIG_MACH_atmega168 is not set
CONFIG_MCU="stm32f103xb"
CONFIG_CLOCK_FREQ=72000000
CONFIG_USBSERIAL=y
CONFIG_SERIAL_PORT=1
CONFIG_FLASH_SIZE=0x10000
CONFIG_RAM_SIZE=0x5000
CONFIG_STACK_SIZE=512
CONFIG_FLASH_START=0x7000
CONFIG_STM32_SELECT=y
CONFIG_MACH_STM32F103=y
# CONFIG_MACH_STM32F405 is not set
# CONFIG_MACH_STM32F407 is not set
# CONFIG_MACH_STM32F446 is not set
CONFIG_MACH_STM32F1=y
CONFIG_HAVE_STM32_USBFS=y
# CONFIG_HAVE_STM32_USBOTG is not set
# CONFIG_STM32_FLASH_START_2000 is not set
CONFIG_STM32_FLASH_START_7000=y
# CONFIG_STM32_FLASH_START_0000 is not set
CONFIG_CLOCK_REF_8M=y

#
# USB ids
#
CONFIG_USB_VENDOR_ID=0x2341
CONFIG_USB_DEVICE_ID=0xabcd
CONFIG_USB_SERIAL_NUMBER="12345"
CONFIG_HAVE_GPIO=y
CONFIG_HAVE_GPIO_ADC=y
CONFIG_HAVE_GPIO_SPI=y
# CONFIG_HAVE_GPIO_I2C is not set
# CONFIG_HAVE_GPIO_HARD_PWM is not set
CONFIG_HAVE_GPIO_BITBANGING=y
CONFIG_INLINE_STEPPER_HACK=y
```

make command...
```
  Build Kconfig config file
#
# configuration written to /opt/klipperlatest/klipper/.config
#
  Creating symbolic link out/board
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/spi_software.o
  Compiling out/src/neopixel.o
  Compiling out/src/stm32/main.o
  Compiling out/src/stm32/watchdog.o
  Compiling out/src/stm32/gpio.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_timer.o
  Compiling out/src/../lib/stm32f1/system_stm32f1xx.o
  Compiling out/src/stm32/stm32f1.o
  Compiling out/src/stm32/adc.o
  Compiling out/src/stm32/spi.o
  Compiling out/src/stm32/usbfs.o
  Compiling out/src/generic/usb_cdc.o
  Building out/compile_time_request.o
Version: v0.7.0-773-gc930fc3-20190820_090132-octopus
  Assembling out/src/../lib/stm32f1/gcc/startup_stm32f103xb.o
  Preprocessing out/stm32.ld
  Linking out/klipper.elf
  Creating hex file out/klipper.bin
```

dmesg output
```
[597867.420322] usb 3-3: new full-speed USB device number 45 using ohci-pci
[597867.580306] usb 3-3: device descriptor read/64, error -62
[597867.848314] usb 3-3: device descriptor read/64, error -62
[597868.140354] usb 3-3: new full-speed USB device number 46 using ohci-pci
[597868.328335] usb 3-3: device descriptor read/64, error -62
[597868.596338] usb 3-3: device descriptor read/64, error -62
[597868.704411] usb usb3-port3: attempt power cycle
[597869.176339] usb 3-3: new full-speed USB device number 47 using ohci-pci
[597869.592388] usb 3-3: device not accepting address 47, error -62
[597869.748357] usb 3-3: new full-speed USB device number 48 using ohci-pci
[597870.192357] usb 3-3: device not accepting address 48, error -62
[597870.192495] usb usb3-port3: unable to enumerate USB device
[597875.312435] usb 3-3: new full-speed USB device number 49 using ohci-pci
[597875.472320] usb 3-3: device descriptor read/64, error -32
[597875.740331] usb 3-3: device descriptor read/64, error -32
[597876.004331] usb 3-3: new full-speed USB device number 50 using ohci-pci
[597876.164451] usb 3-3: device descriptor read/64, error -32
[597876.432456] usb 3-3: device descriptor read/64, error -32
[597876.540544] usb usb3-port3: attempt power cycle
[597877.040460] usb 3-3: new full-speed USB device number 51 using ohci-pci
[597877.456461] usb 3-3: device not accepting address 51, error -32
[597877.612462] usb 3-3: new full-speed USB device number 52 using ohci-pci
[597878.028482] usb 3-3: device not accepting address 52, error -32
[597878.028607] usb usb3-port3: unable to enumerate USB device
```

KevinOConnor, 2019-08-20:
A separate user was reporting that if they hit the reset button a couple of times, it would eventually come up.  (Though that user also reports the latest version does boot on its own.)  Do you eventually see a connection when clicking reset?

-Kevin

vladbabii, 2019-08-20:
I will try that but in a couple of hours. Is the reset press required every boot? Because with older firmware it always connects to usb no matter what.

vladbabii, 2019-08-20:
Done a ton of resets. No change. same errors ( -62 ). Tried different cables, different ports, a laptop, a desktop, a pi, a usb 1.1 hub, a usb 3.0 hub....

KevinOConnor, 2019-08-21:
Alas, I'm unable to reproduce the problem, so it's difficult to track this down.

Are you able to successfully compile and run ec3d865 ?  If so, can you provide the klipper log file from that run?

I put some code changes up on the work-stm32f1-20190820 branch:
cd ~/klipper ; git fetch ; git checkout origin/work-stm32f1-20190820 ; make ; ...flash...

You could try with that branch - but the changes on that branch are just wild guesses at this point.

-Kevin

vladbabii, 2019-08-21:
trying ec3d865

```
configuration written to /opt/klipperlatest/klipper/.config

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

blue@octopus:/opt/klipperlatest/klipper$ make
  Build Kconfig config file
  Creating symbolic link out/board
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/spi_software.o
  Compiling out/src/neopixel.o
  Compiling out/src/stm32f1/main.o
  Compiling out/src/stm32f1/gpio.o
  Compiling out/src/stm32f1/adc.o
  Compiling out/src/stm32f1/spi.o
  Compiling out/src/../lib/stm32f1/hal/source/stm32f1xx_ll_adc.o
  Compiling out/src/../lib/stm32f1/hal/source/stm32f1xx_ll_spi.o
  Compiling out/src/../lib/stm32f1/hal/source/stm32f1xx_ll_gpio.o
  Compiling out/src/../lib/stm32f1/hal/source/stm32f1xx_ll_exti.o
  Compiling out/src/../lib/stm32f1/hal/source/stm32f1xx_ll_fsmc.o
  Compiling out/src/../lib/stm32f1/hal/source/stm32f1xx_ll_usb.o
  Compiling out/src/../lib/stm32f1/hal/source/stm32f1xx_ll_dma.o
  Compiling out/src/../lib/stm32f1/hal/source/stm32f1xx_ll_rtc.o
  Compiling out/src/../lib/stm32f1/hal/source/stm32f1xx_ll_pwr.o
  Compiling out/src/../lib/stm32f1/hal/source/stm32f1xx_ll_utils.o
  Compiling out/src/../lib/stm32f1/hal/source/stm32f1xx_ll_i2c.o
  Compiling out/src/../lib/stm32f1/hal/source/stm32f1xx_ll_rcc.o
  Compiling out/src/../lib/stm32f1/hal/source/stm32f1xx_ll_dac.o
  Compiling out/src/../lib/stm32f1/hal/source/stm32f1xx_ll_crc.o
  Compiling out/src/../lib/stm32f1/hal/source/stm32f1xx_ll_usart.o
  Compiling out/src/../lib/stm32f1/hal/source/stm32f1xx_ll_sdmmc.o
  Compiling out/src/../lib/stm32f1/hal/source/stm32f1xx_ll_tim.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_timer.o
  Compiling out/src/../lib/stm32f1/system_stm32f1xx.o
  Compiling out/src/stm32f1/usbserial.o
  Compiling out/src/generic/usb_cdc.o
  Building out/compile_time_request.o
Version: v0.7.0-733-gec3d865-20190821_090759-octopus
  Assembling out/src/../lib/stm32f1/gcc/startup_stm32f103xb.o
  Preprocessing out/stm32f1.ld
  Linking out/klipper.elf
  Creating hex file out/klipper.bin
```

config file
```
#
# Automatically generated file; DO NOT EDIT.
# Klipper Firmware Configuration
#
# CONFIG_LOW_LEVEL_OPTIONS is not set
# CONFIG_MACH_AVR is not set
# CONFIG_MACH_ATSAM is not set
# CONFIG_MACH_ATSAMD is not set
# CONFIG_MACH_LPC176X is not set
# CONFIG_MACH_STM32F0 is not set
CONFIG_MACH_STM32F1=y
# CONFIG_MACH_STM32F4 is not set
# CONFIG_MACH_PRU is not set
# CONFIG_MACH_LINUX is not set
# CONFIG_MACH_SIMU is not set
CONFIG_STEP_DELAY=2
CONFIG_BOARD_DIRECTORY="stm32f1"
CONFIG_CLOCK_FREQ=72000000
CONFIG_USBSERIAL=y
CONFIG_FLASH_START=0x7000
CONFIG_STM32F1_SELECT=y
# CONFIG_STM_FLASH_START_2000 is not set
CONFIG_STM_FLASH_START_7000=y
# CONFIG_STM_FLASH_START_0000 is not set

#
# USB ids
#
CONFIG_USB_VENDOR_ID=0x2341
CONFIG_USB_DEVICE_ID=0xabcd
CONFIG_USB_SERIAL_NUMBER="12345"
CONFIG_HAVE_GPIO=y
CONFIG_HAVE_GPIO_ADC=y
CONFIG_HAVE_GPIO_SPI=y
# CONFIG_HAVE_GPIO_I2C is not set
# CONFIG_HAVE_GPIO_HARD_PWM is not set
CONFIG_HAVE_GPIO_BITBANGING=y
CONFIG_INLINE_STEPPER_HACK=y
```

vladbabii, 2019-08-21:
work-stm32f1-20190820 branch

```
  Build Kconfig config file
  Creating symbolic link out/board
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/spi_software.o
  Compiling out/src/neopixel.o
  Compiling out/src/stm32/main.o
  Compiling out/src/stm32/watchdog.o
  Compiling out/src/stm32/gpio.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_timer.o
  Compiling out/src/../lib/stm32f1/system_stm32f1xx.o
  Compiling out/src/stm32/stm32f1.o
  Compiling out/src/stm32/adc.o
  Compiling out/src/stm32/i2c.o
  Compiling out/src/stm32/spi.o
  Compiling out/src/stm32/usbfs.o
  Compiling out/src/generic/usb_cdc.o
  Building out/compile_time_request.o
Version: v0.7.0-779-gd2118c8-20190821_091559-octopus
  Assembling out/src/../lib/stm32f1/gcc/startup_stm32f103xb.o
  Preprocessing out/stm32.ld
  Linking out/klipper.elf
  Creating hex file out/klipper.bin
```

config:
```
#
# Automatically generated file; DO NOT EDIT.
# Klipper Firmware Configuration
#
# CONFIG_LOW_LEVEL_OPTIONS is not set
# CONFIG_MACH_AVR is not set
# CONFIG_MACH_ATSAM is not set
# CONFIG_MACH_ATSAMD is not set
# CONFIG_MACH_LPC176X is not set
CONFIG_MACH_STM32=y
# CONFIG_MACH_STM32F0 is not set
# CONFIG_MACH_PRU is not set
# CONFIG_MACH_LINUX is not set
# CONFIG_MACH_SIMU is not set
CONFIG_STEP_DELAY=2
CONFIG_BOARD_DIRECTORY="stm32"
CONFIG_MCU="stm32f103xb"
CONFIG_CLOCK_FREQ=72000000
CONFIG_USBSERIAL=y
CONFIG_SERIAL_PORT=1
CONFIG_FLASH_SIZE=0x10000
CONFIG_RAM_SIZE=0x5000
CONFIG_STACK_SIZE=512
CONFIG_FLASH_START=0x7000
CONFIG_STM32_SELECT=y
CONFIG_MACH_STM32F103=y
# CONFIG_MACH_STM32F405 is not set
# CONFIG_MACH_STM32F407 is not set
# CONFIG_MACH_STM32F446 is not set
CONFIG_MACH_STM32F1=y
CONFIG_HAVE_STM32_USBFS=y
# CONFIG_HAVE_STM32_USBOTG is not set
# CONFIG_STM32_FLASH_START_2000 is not set
CONFIG_STM32_FLASH_START_7000=y
# CONFIG_STM32_FLASH_START_0000 is not set
CONFIG_CLOCK_REF_8M=y

#
# USB ids
#
CONFIG_USB_VENDOR_ID=0x2341
CONFIG_USB_DEVICE_ID=0xabcd
CONFIG_USB_SERIAL_NUMBER="12345"
CONFIG_HAVE_GPIO=y
CONFIG_HAVE_GPIO_ADC=y
CONFIG_HAVE_GPIO_SPI=y
CONFIG_HAVE_GPIO_I2C=y
# CONFIG_HAVE_GPIO_HARD_PWM is not set
CONFIG_HAVE_GPIO_BITBANGING=y
CONFIG_INLINE_STEPPER_HACK=y
```

vladbabii, 2019-08-21:
Should i try revisions after ec3d865 until i find one that does not work?

vladbabii, 2019-08-21:
I will try building a couple more firmware versions after the one mentioned above and see what happens.

vladbabii, 2019-08-21:
ec3d865b517affd77678e5b1a45ef4691619726d works
but the next one
8b9cc62359057a686929cc713ffe2931e2203946 does not work

vladbabii, 2019-08-21:
@KevinOConnor  can i use some other 2 pins for serial communication instead of usb port? I think that might solve things temporarily if the USB connection is the only issue.

vladbabii, 2019-08-21:
I got some files from another user on discord that uses this fork 

Git: https://github.com/BlackStump/klipper
Revision: b2eaab922621f700d52b2519284eb0299aad010e

[config.menu.zip](https://github.com/KevinOConnor/klipper/files/3524529/config.menu.zip)
[klipper.zip](https://github.com/KevinOConnor/klipper/files/3524531/klipper.zip)

This works on my board

vladbabii, 2019-08-21:
compiling on a pi 3b+ and on a ubuntu 18.04 produce different files for same git revision

[image]
[image]

KevinOConnor, 2019-08-21:
> can i use some other 2 pins for serial communication instead of usb port?

The mini has a "tft" header that seems to expose uart pins.  So, you should be able to disable USB support in "make menuconfig" and use a regular serial port instead.  If you do this, be sure the voltages match correctly.

> Files have different size. I was expecting the same identical size for both sources

Different versions of gcc produce slightly different code.  It's normal.

-Kevin

vladbabii, 2019-08-21:
@KevinOConnor compiling for skr mini works only on raspberry pi or arm systems. I'll leave this ever if anyone happens to have the same issue. 

I think for now we can close this ticket? Or should I leave it open due to different behaviour of firmware depending on where it was compiled?

KevinOConnor, 2019-08-21:
Can you provide more details?  Are you now able to boot the latest code on the mini?  What OS is unable to correctly compile Klipper?  Is that system able to compile an older version of Klipper?

-Kevin

vladbabii, 2019-08-21:
This system cannot make workable firmwares:

```
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.3 LTS
Release:	18.04
Codename:	bionic
```

On this pi builds work:
```
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.4 (stretch)
Release:	9.4
Codename:	stretch
```

vladbabii, 2019-08-21:
And yes, latest klipper can be built on the pi and it works on skr mini. I have another pi that cannot build working firmware but the os is similar and fully upgraded. I will try to make a diff between the two pi's in the next 2-3 days when I have time.

vladbabii, 2019-08-21:
I also noticed that firmware restart doesn't work right on newer firmwares, i have to restart host

```
Internal error during connect: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
```

AshIzat, 2019-08-22:
I believe I am running in to the same issue on the skr mini e3 (same mcu as the mini). I'll check dmesg output when I get home (8 hours or so) but the symptoms are that the device is visible in /dev when it is first plugged in, but disappears as soon as klipper tries to connect, and the only way to get it to reappear is to unplug and replug the usb cable.

Steps take to get to this point: After a fresh clone of master, I chose stm32f103 with 28kb bootloader in menuconfig, ran make and copied out/klipper.bin onto the sdcard and inserted it into the board.

I'd be happy to help in any debugging efforts with this board.

AshIzat, 2019-08-22:
Reading about the possible differences in build environment I'll add that this is all taking place on a Pi 3B (not 3B+), I don't know if that is relevant or not.

AshIzat, 2019-08-22:
I've been doing some reading around and my issue may be unrelated to this after all.

People have been reporting similar symptoms to mine with Marlin and it turns out the MCU needs to set PC13 to output low at startup to properly initialise the USB connection, this is mentioned
in [this comment](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/issues/2#issuecomment-513149032) on the the SKR mini e3 github, and fixed in [PR  #14679](https://github.com/MarlinFirmware/Marlin/pull/14679) in marlin.

I believe I should be able to set this correctly in menuconfig, so I'll test and report back in a few hours.

vladbabii, 2019-08-22:
@AshIzat how did you set pin PC13 low? I will try on my mini

vladbabii, 2019-08-23:
@KevinOConnor can you point me in the right direction to set PC13 low on startup in firmware?

AshIzat, 2019-08-23:
@vladbabii as far as I can tell, the  SKR Mini 1.1 doesn't have the same issue as the Mini E3, on the Mini E3 the D+ line of the USB connection has it's pull-up resistor connected via a p-channel mosfet controlled by PC13. I don't have an SKR Mini 1.1 so I can't be sure, but looking at pictures it seems to have it's pull-up connected directly (I don't see anything that looks like a mosfet near the USB connection). Even if it does have the same problem, it wouldn't be PC13 that needs setting low, as that appears to be used in the expansion header.

Sorry for causing confusion, I thought my problem was the same as in this issue (similar symptoms, same mcu), but it turns out it wasn't, once I set PC13 low on startup USB worked fine for me on the Mini E3.

For the reference of anyone reading this who does have the Mini E3, when you run `make menuconfig` before building the firmware, you can set the initial states of pins at MCU startup, `!PC13` is the value you need to bring PC13 Low at startup.

vladbabii, 2019-09-02:
@KevinOConnor If it helps, I can give you ssh access to a VM with whatever ubuntu flavor you want.

pusewicz, 2019-09-04:
I'm having the same problem here. Plugging in the device with dmesg:

```
[ 4188.792809] usb 6-1: new full-speed USB device number 40 using ohci-platform
[ 4188.984812] usb 6-1: device descriptor read/64, error -32
[ 4189.284819] usb 6-1: device descriptor read/64, error -32
[ 4189.580821] usb 6-1: new full-speed USB device number 41 using ohci-platform
[ 4189.772821] usb 6-1: device descriptor read/64, error -32
[ 4190.072829] usb 6-1: device descriptor read/64, error -32
[ 4190.180897] usb usb6-port1: attempt power cycle
[ 4190.684848] usb 6-1: new full-speed USB device number 42 using ohci-platform
[ 4191.100855] usb 6-1: device not accepting address 42, error -32
[ 4191.288858] usb 6-1: new full-speed USB device number 43 using ohci-platform
[ 4191.704853] usb 6-1: device not accepting address 43, error -32
[ 4191.704937] usb usb6-port1: unable to enumerate USB device
```

vladbabii, 2019-09-04:
Compiled on Linux or pi?

pusewicz, 2019-09-04:
Compiled on OrangePi Zero running `Welcome to Ubuntu Bionic with Armbian Linux 4.19.57-sunxi`.

pusewicz, 2019-09-04:
Funny thing is that it worked originally, but I can't remember what I changed.

pusewicz, 2019-09-04:
@KevinOConnor What can I do to help out?

vladbabii, 2019-09-05:
@pusewicz if you give me your compile settings and git version i can try compiling a file on a pi and check if that works or not...

pusewicz, 2019-09-05:
@vladbabii 

## .config

```
#
# Automatically generated file; DO NOT EDIT.
# Klipper Firmware Configuration
#
CONFIG_LOW_LEVEL_OPTIONS=y
# CONFIG_MACH_AVR is not set
# CONFIG_MACH_ATSAM is not set
# CONFIG_MACH_ATSAMD is not set
# CONFIG_MACH_LPC176X is not set
CONFIG_MACH_STM32=y
# CONFIG_MACH_STM32F0 is not set
# CONFIG_MACH_PRU is not set
# CONFIG_MACH_LINUX is not set
# CONFIG_MACH_SIMU is not set
CONFIG_STEP_DELAY=2
CONFIG_BOARD_DIRECTORY="stm32"
CONFIG_MCU="stm32f103xb"
CONFIG_CLOCK_FREQ=72000000
CONFIG_USBSERIAL=y
CONFIG_SERIAL_PORT=1
CONFIG_FLASH_START=0x8007000
CONFIG_FLASH_SIZE=0x10000
CONFIG_RAM_START=0x20000000
CONFIG_RAM_SIZE=0x5000
CONFIG_STACK_SIZE=512
CONFIG_STM32_SELECT=y
CONFIG_MACH_STM32F103=y
# CONFIG_MACH_STM32F405 is not set
# CONFIG_MACH_STM32F407 is not set
# CONFIG_MACH_STM32F446 is not set
CONFIG_MACH_STM32F1=y
CONFIG_HAVE_STM32_USBFS=y
# CONFIG_HAVE_STM32_USBOTG is not set
# CONFIG_STM32_FLASH_START_2000 is not set
CONFIG_STM32_FLASH_START_7000=y
# CONFIG_STM32_FLASH_START_0000 is not set
CONFIG_STM32_CLOCK_REF_8M=y
# CONFIG_STM32_CLOCK_REF_INTERNAL is not set
CONFIG_CLOCK_REF_8M=y

#
# USB ids
#
CONFIG_USB_VENDOR_ID=0x2341
CONFIG_USB_DEVICE_ID=0xabcd
CONFIG_USB_SERIAL_NUMBER="12345"
# CONFIG_CUSTOM_STEP_DELAY is not set
CONFIG_INITIAL_PINS=""
CONFIG_HAVE_GPIO=y
CONFIG_HAVE_GPIO_ADC=y
CONFIG_HAVE_GPIO_SPI=y
CONFIG_HAVE_GPIO_I2C=y
# CONFIG_HAVE_GPIO_HARD_PWM is not set
CONFIG_HAVE_GPIO_BITBANGING=y
CONFIG_INLINE_STEPPER_HACK=y
```

SHA1: 12feb6d7fbd85d34a81d23e59472e3083622a206

vladbabii, 2019-09-05:
[klipper.zip](https://github.com/KevinOConnor/klipper/files/3579203/klipper.zip)
@pusewicz here you go

pusewicz, 2019-09-05:
@vladbabii Did you try it?

anm8tor, 2019-09-06:
I have a Pi 4 and SKR Mini 1.1
My current problem is that I get no recognition whatsoever after plugging in the USB cable. Not on Klipper yet, just trying to get Octoprint connected. "lsusb" produces this, both before and after pluggin in the USB cable:

pi@octopi:~ $ lsusb
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@octopi:~ $ lsusb
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

vladbabii, 2019-09-06:
@pusewicz - I built the image for you to try. I don't have anything free at the moment to use it on

vladbabii, 2019-09-06:
@anm8tor - try the firmware I uploaded above, it should work

anm8tor, 2019-09-06:
@vladbabii This image replaces both Klipper and Octopi on the Pi4?

vladbabii, 2019-09-06:
That's klipper firmware for the board

KevinOConnor, 2019-09-06:
I got a chance to compile Klipper on an Ubuntu 18.04 VM.  However, I was unable to reproduce the problem.  I compiled for stm32f103 (with an 8KiB bootloader) and the resulting binary came up fine on my "blue pill" test board.

Could you verify the problem is still present with the very latest Klipper version?

-Kevin

BlackStump, 2019-09-07:
I have also lost the usb when updating firmware on this board. if put firmware.bin from a few weeks ago it finds the usb again but with the latest pull and compiled firmware.bin there is no serial in /dev
firmware compiled on beaglebone black debian.

anm8tor, 2019-09-07:
@vladbabii Forgive my ignorance, this is still very new to me. My understanding was that I need Octoprint/Octopi and Klipper installed together. Are you saying I can just install the Killer image?

BlackStump, 2019-09-07:
@anm8tor 
https://octoprint.org/
Octoprint is a web interface for 3d printers and not the only web interface that can be used with klipper just the web interface of choice by the dev.

Octopi is a image for the rasberrypi sbc that has octoprint already installed.

Klipper is software/firmware for 3d printer they are not the same and not dependent on each other but can be used together . 

have you read the install process?
https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md

anm8tor, 2019-09-07:
@blackstump Thanks for the response. I haven't read the entire thing, but the first line in the file is what I was referencing. I originally commented on this thread, because my Pi4 with the following Octopi image doesn't see my SKR mini at all, but sees other USB devices just fine. 

https://storage.googleapis.com/octoprint/2019-06-24_2019-06-20-octopi-buster-lite-0.17.0.zip

vladbabii, 2019-09-09:
@anm8tor 
You can run
```
dmesg -c
```

to clear all message logs from pi, then plugin the usb cable then run
```
dmesg
```
and you can get the logs from the pi of what the pi sees when the board gets plugged in. You can attach the log here or copy-paste it so we can see what error you get.

Also, try using the reset button on the board after flashing.

vladbabii, 2019-09-09:
@anm8tor i will try later today the firmware image i sent you - i need to buy a new microsd card, the last one died on me and i only have 128gb ones that won't work with the skr.

KevinOConnor, 2019-09-10:
Alas, I'm unable to reproduce this failure.  If this is still an issue, please checkout a known good version of Klipper (`git checkout <revision>`), compile it, load it on the board, obtain the klipper log, grab the out/ directory (`tar cfz klipper-good.tar.gz out/`).  Then checkout the latest version of Klipper (`git checkout origin/master`), compile it on identical machine and compiler, load it on the board, grab the klipper log, grab the out/ directory (`tar cfz klipper-bad.tar.gz out/`).  Finally, attach all four files here along with the OS revision (`cat /etc/os-release`) and compiler version (`arm-none-eabi-gcc --version ; arm-none-eabi-ld --version`).

-Kevin

[66 more replies]