Klipper3d/klipper #1885: Subdirectories on a virtual SD card

mbgroot, 2019-08-13:
Is it possible to implement support for subdirectories on a virtual SD card?

Fake klipper log for bot
[klippy.log](https://github.com/KevinOConnor/klipper/files/3496815/klippy.log)

KevinOConnor, 2019-08-17:
I'm sure it's possible, but I do not think there is a common mechanism for doing that in g-code.  FWIW, I'm leery of adding non-standard "enhancements" to the legacy g-code commands.

-Kevin

KevinOConnor, 2019-08-17:
To be clear, one could certainly add new "extended g-code" commands to do that if one wished to.

-Kevin

mbgroot, 2019-08-18:
Why do you need G-codes to print from subdirectories? It is not solved at the code level?

KevinOConnor, 2019-09-14:
I'm going to close this issue as I believe the original question was answered (it is possible but someone would need to code it up).

-Kevin

closed 2019-09-14 by KevinOConnor