Klipper3d/klipper #1886: No section: 'printer' Please Help me

mapa3m1, 2019-08-13:
I got this error  
[klippy.log](https://github.com/KevinOConnor/klipper/files/3498054/klippy.log)
" No section: 'printer' "
16:54:55 Connected to host via /tmp/printer @250000bps
16:54:55 Firmware version: v0.7.0-755-geffe6f6
16:55:00 No section: 'printer'
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
Klipper state: Not ready
16:55:00 No section: 'printer'
16:55:00 No section: 'printer'

Hywelmartin, 2019-08-13:
https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md#configuring-klipper

timmit99, 2019-08-15:
Mapa, can you post your config file? It seems like you are kissing the "printer" sections which defines the type of machine and other core definitions