Klipper3d/klipper #1887: [FR] extruder_calibrate helper

StefanCohen, 2019-08-14:
Correct calibration of step_distance for the extruder is essential for print quality and  a prerequisite for properly tuning Pressure Advance. There are plenty of ways to do it but it would be helpful to get some assistance as it is tedious (but something that really should be done every time you change filament).  Klipper also uses a different unit for Esteps compared to other slicers, so assistance with tuning could potentially reduce support burden. A helper process with associated config section could look like this: 

Config:

```
[extruder_calibrate]
hotend_temp: 200 # Hotend temperature in C during calibration (required).
toolhead_position: 50,50,50 # X, Y, Z position in mm where to move the nozzle during the calibration to make room for and catch extruded filament. Default: 50,50,50.
extrude_speed: 100 #How fast to extrude during calibration in mm/minute. 
```

**Meta code for helper: EXTRUDER_CALIBRATE** 

- User submits extended g_code: EXTRUDER_CALIBRATE
- Verify presence of [extruder_calibrate] section in config else return error. 
- Ask user to mark filament 120mm from extruder and START_CALIBRATION or ABORT.
- Heat hotend to hotend_temp 
- Home all axxis
- Move nozzle to toolhead_position 
- Wait for hotend to reach hotend_temp
- Reset the “extruded material”: G92 E0
- Extrude 2 * 50mm filament at extrude_speed feedrate (to account for default 50mm max_extrude_only_distance limit. Alternativelly a forced move of a single 100mm extrusion)
- Disable hotend heater.
- Ask user to input measured remaining filament to mark in mm.
- Get current step_distance from printer.
- Calculate new step_distance: new_step_distance = ((120 - remaining_filament ) / 100) * current_step_distance
- return new_step_distance to user and ask to set and store new value in config file & restart or abort.
- Execute SAVE_CONFIG & RESTART or ABORT

I have tried to implement the above routine using a custom macro but it doesn’t seem to be possible to get or set step_distance using the “printer” pseudo variable. I have tried with printer.extruder.step_distance and printer.toolhead.step_distance without success and I can’t find any detailed documentation about the object model. It looks like the only available option is to implement the above as a new host module. I’m unfortunately not savvy enough to implement it myself without extensive assistance but I would be happy to test & document it if someone would like to spend some time on this. A minimalistic alternative could be a custom macro if someone could help with a way to at least get the current step_distance from the printer variable so a new step_distance can be returned to the user and then manually entered into the printer.conf.

A nice addition would be a default menu under __prepare to start calibration, enter measured remaining filament using encoder and save&restart or abort configuration. Default is nice, but it would be trivial to implement as a custom menu once the above helper is available.

Example:

```
[menu __prepare]
type: list
enable: !toolhead.is_printing
name: Prepare
items:
    .__extruder_tuning #New entry for Extruder Calibration
    .__delta_calib
    .__bedprobe
    .__hotend_pid_tuning
    .__hotbed_pid_tuning
    .__host_restart
    .__firmware_restart

[menu __prepare __extruder_tuning]
type: list
enable: !toolhead.is_printing
name: Mesh Bed Level
items:
    .start #Initiate calibration
    .remaining_filament #Enter measured remaining filament using encoder
    .result #display new extruder step_distance
    .abort #abort calibration without saving
    .save #SAVE_CONFIG & RESTART

```

Fake klipper log for bot:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3500971/klippy.log)

StefanCohen, 2019-08-14:
Never mind. I managed to write it myself after all. I'll clean it up and will send a pull request for review when I have figured out how to do that.

jakep82, 2019-08-15:
Step_distance should be configured once, and each filament should be calibrated in the slicer.  Changing step_distance every time I switch filament would be an awful user experience compared with simply selecting the filament I'm using my slicer which I've already configured with flow percentage and pressure_advance values.

AndyWo83, 2019-08-27:
Hello @StefanCohen 
the extruder must be calibrated with nothing connected to it. 
There must be no back pressure on the extruder with the nozzle. 
The rest you can set in Prusa Slicer on the Extrusion Multiplier and the Pressure Advance value you can enter at Filament Start GCode, so you would have for each filament its own setting. 
With Cura, you'd have to fix that via the flow, and you could add the pressure advance values via Linear Advance when adding in printer.cfg 
[gcode_macro M900] 
gcode: SET_PRESSURE_ADVANCE ADVANCE = {K} #Linear Advance Value from Cura or other Slicers

StefanCohen, 2019-08-27:
Thanks for the feedback . I'll close this issue.

closed 2019-08-27 by StefanCohen