Klipper3d/klipper #1888: BLTouch offset not respected in bed leveling

wormiedk, 2019-08-14:
I still find that the bed leveling does not respect the offsets for the BLTouch. 

In my config I have set BLTOUCH offset to:
```
[bltouch]
x_offset: 37
y_offset: 2
```

However in order to get Klipper to measure on the whole build plate (200x200) I have to set limits for bed leveling to
```
[bed_mesh]
speed: 80
horizontal_move_z: 5
min_point: 0,30
max_point: 160,160
```

If I increase max point to 200 klipper tries to probe outside the bed. Further I should not be able to set the min X to less than 37. 
When I perform the probing the display shows X=0 when the probe is actually at X=37 see attached photos.
[image]
[image]

[config.txt](https://github.com/KevinOConnor/klipper/files/3501379/config.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/3501380/klippy.log)
[printer.txt](https://github.com/KevinOConnor/klipper/files/3501381/printer.txt)

Hywelmartin, 2019-08-14:
https://github.com/KevinOConnor/klipper/blob/effe6f6ddd5132db4ab4f88bf8616aab575e7691/config/example-extras.cfg#L171

jakep82, 2019-08-15:
@Hywelmartin linked the relevant config section.  Probe offsets are not considered when defining probe boundaries.  Everything is based on nozzle location, so you need to do the math yourself when creating probe boundaries.

sennalb, 2019-09-01:
If "Everything is based on nozzle location" what is the purpose of defining the x_offset in [bltouch] section ?
Thanks

KevinOConnor, 2019-09-06:
The documentation does describe this (specifically in config/example-extras.cfg).  The bed region to probe is determined in nozzle coordinates (not coordinates of the probe).  Once the bed region is measured, the probe x/y_offset is used to correct for any offsets.

It is understood that this has been confusing for users and I understand that Arksine is looking at alternative ways to configure the bed region to probe in bed_mesh.  Lets track this in #1862 . 

-Kevin

closed 2019-09-14 by KevinOConnor