Klipper3d/klipper #1889: Endstop a still triggered after retract

JeepersMD, 2019-08-14:
Good afternoon,
I have finally got things another few steps and actually thru the config checks got the stepper motors to move.  But they will not home at all.  I am using a Bigtreetech SKR 1.3 board and had to take the 2130's out and go back on my Anycubic delta with the old chipset A4988 (long story), but this got movement.  But it stops after trying with the above error message.  It bumps a second and then done. What's my next move?
Look forward to hearing from you.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3503441/klippy.log)

jakep82, 2019-08-15:
Have you verified your endstops?

https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md#verify-endstops

JeepersMD, 2019-08-15:
All report triggered.

jakep82, 2019-08-15:
Okay, then the directions at the link explain how to fix that.

Hywelmartin, 2019-08-16:
If the endstop appears inverted (it reports "open" when triggered and vice-versa) then add a "!" to the pin definition (for example, "endstop_pin: ^!ar3"), or remove the "!" if there is already one present.

KevinOConnor, 2019-09-06:
I'm going to close this issue as it appears the original question was answered (follow the directions in the documentation).

-Kevin

closed 2019-09-06 by KevinOConnor