Klipper3d/klipper #1890: No display no control

cinealfa, 2019-08-14:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3503546/klippy.log)

I know it is probably something I am doing wrong but I can't seem to figure out how to get Klipper to work. the display is no functional and the printer seems to be partially connected but I can figure out what I am doing wrong.
The board is an MKS gen l and it does function with marlin 2.0

BlackStump, 2019-08-15:
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

Hywelmartin, 2019-08-15:
error: Unable to connect
Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer
Klipper state: Not ready
Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer

Unable to open port: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'

so this setting is faulty

```
[mcu]
serial = /dev/ttyACM0  <---
pin_map = arduino
```

cinealfa, 2019-08-15:
@Hywelmartin your help was greatly appreciated there were two different serial connections in the config. thank you for actually taking the time to communicate clearly what the issue was.

closed 2019-08-15 by cinealfa