Klipper3d/klipper #1893: Klipper Update on Ubuntu

novio8, 2019-08-17:
Hi,
today I decided to update Klipper that was succesfully running on Ubuntu 18.04 machine.
I followed instructions and after restart I couldn't connect to 3D printer, and display is blank.
I'v run the command: sudo service klipper status and the response is bellow.

 klipper.service - Starts klipper on startup
   Loaded: loaded (/etc/systemd/system/klipper.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2019-08-17 16:42:10 CEST; 4min 45s ago
  Process: 946 ExecStart=/home/ubunta/klippy-env/bin/python /home/ubunta/klipper/scripts/klipper/klippy/klipp
 Main PID: 946 (code=exited, status=217/USER)

avg 17 16:42:10 ubunta-XH61V systemd[1]: Started Starts klipper on startup.
avg 17 16:42:10 ubunta-XH61V systemd[946]: klipper.service: Failed to determine user credentials: No such pro
avg 17 16:42:10 ubunta-XH61V systemd[946]: klipper.service: Failed at step USER spawning /home/ubunta/klippy-
avg 17 16:42:10 ubunta-XH61V systemd[1]: klipper.service: Main process exited, code=exited, status=217/USER
avg 17 16:42:10 ubunta-XH61V systemd[1]: klipper.service: Failed with result 'exit-code'.

I don't know what mistake have I done so any help is much appreciated.

Regards
Mario

novio8, 2019-08-23:
I tried to find klipper log file, but there is none.  Please advise how to proceed
best regards
Mario

BlackStump, 2019-08-23:
what script did you use to install klipper on a ubuntu system?
ie did you use
./klipper/scripts/install-ubuntu-18.04.sh

novio8, 2019-08-23:
BlackStump,
I did use ./klipper/scripts/install-ubuntu-18.04.sh

BlackStump, 2019-08-23:
my guess is you have a path issue to where you installed klipper and also possible user account issue
this is in the ubuntu install script
# This script installs Klipper on an Ubuntu 18.04 machine with Octoprint
# Assumes the user running this is named "octoprint"

novio8, 2019-08-24:
BlackStump, 
I reinstalled Klipper and changed user name in ./klipper/scripts/install-ubuntu-18.04.sh
thank you

closed 2019-08-24 by novio8