Klipper3d/klipper #1894: Chaining RPis for consistent performance

Asterchades, 2019-08-18:
I don't know if this counts as an OctoPrint/Pi thing or a Klipper thing, so I'll start here and see where it takes me.

I've been playing around with Klipper the last couple of days and rather liking it. The results on an Ender-2 (ATMega1284p @ 16MHz) as compared to leaving it handle everything are truly extraordinary - far beyond what I would have expected simply by changing the host controller. I've encountered a bizarre, persistent "SD write not supported" message which I've managed to work around, but that's been the only issue which doesn't seem to have an explanation (and I haven't exhausted my personal troubleshooting on that one yet).

What I have found, however, is that my RPi 2 is not adequate (for what I'm currently printing, at least) to run both OctoPrint _and_ Klipper. The print can and does pause for seconds at a time every layer or two, which will invariably have an effect on the resulting print. Using the Virtual SD card feature has allowed me to get around this limitation but also suppresses most of the OctoPrint functionality (GCode viewer, object/area cancellation, etc) , which leaves me wondering: can you chain two RPis together to have both OctoPrint and Klipper with their own dedicated hardware?

By that, I mean could I get a second RPi (an RPi Zero or an older second hand unit) to run the actual Klipper host software, which is then controlled by my existing RPi 2 running OctoPrint? This would be cheaper than replacing the RPi 2 with an RPi 4, while also alleviating any issue with the two pieces of software contending for system time and resources even on the more powerful hardware.

I'm mainly assuming this is more a Klipper thing as it already runs virtual ports. I feel they'll likely be required for what I'm hoping to do.

danielfmo, 2019-08-18:
Hi,

I believe you're trying to came up with a fix for a problem you do not fully understand. 
The shuttering problem is caused by Octoprint communication layer alone, it's a subject discussed for long and promised to be solved in the next major release (1.4.0). 

The issue is due to the bact that Octoprint is not able to properly buffer GCODE files with high density commands. If you're using CURA set the Maximun Resolution parameter to at least 0.05, I use 0.1.

Asterchades, 2019-08-18:
I've no doubt that I don't fully understand the problem. After all, I've been trying to figure out why I couldn't write to the Virtual SD card... which is by design, and documented.

I was merely looking at the situation as an outsider looking in. To me, it seemed like one RPi2 could run OctoPrint _or_ Klipper without issue, but not both. Thus the simple solution would appear to be using a second RPi so you can have one doing each task. Granted it's not nearly that simple (it never is, is it?) but that's how it looks from the outside.

Admittedly my test model was something of a torture test (the base model has over a quarter of a million vertices). When left to do all its own work, the ATMega could at best give only reasonable results when almost every advanced feature was turned off, running at half the maximum speed, 1/3 the acceleration, and a 0.2 maximum resolution. And even then it looked very wind-swept.

I am, however, very pleased to hear that it's an issue which should be resolved without massive amounts of screwing around on my part or extra purchases. Cheers for the information!

closed 2019-08-18 by Asterchades