Klipper3d/klipper #1898: skr_mini replicape

BlackStump, 2019-08-20:
I am using multi mcu.
 Replicape as the main and Skr mini for Z steppers and it seemed to be working fine, klipper was last pulled last month so I wanted to update and try some of the newest features 
but after the lasted pull it seems to have broken something, I cant get it working at all. if I revert back to last months image it works fine again.

lastest that has problems
[klippy_latest_broken.log](https://github.com/KevinOConnor/klipper/files/3518597/klippy_latest_broken.log)

working log
[klippy.log](https://github.com/KevinOConnor/klipper/files/3518598/klippy.log)

KevinOConnor, 2019-08-20:
You'll need to "flash" the linux mcu code (make menuconfig ; select linux... ; make flash).  Note, though, that there is an issue with the skr mini with the latest code - #1884 

-Kevin

BlackStump, 2019-08-20:
when I did the update I did pru first, linux second and skr mini last,
are you saying I need to do the linux as the last step?
in any case I wait until the skr mini code is sorted before trying to update again

with the older image I did the skr mini last as the pru/linux was already done.

BlackStump, 2019-08-20:
I have just done a clean image and it is working with the latest pull

closed 2019-08-20 by BlackStump

KevinOConnor, 2019-08-20:
So, the skr mini came up for you without issue on the latest Klipper code?

-Kevin

BlackStump, 2019-08-20:
@KevinOConnor 
yep working fine no problems so far, my problem was user error I think when making the pru/linux firmware I was using "make" instead of "make flash"
It is booting fine and homing fine will spend some more time today on it

BlackStump, 2019-08-21:
there does not seem any problem with bringing the skr mini up
seems at this stage reliable, printing atm