Klipper3d/klipper #1899: MCU 'mcu' shutdown: Timer too close on long prints

darkernoise, 2019-08-20:
MCU 'mcu' shutdown: Timer too close occurring on long prints.  I have had the same print fail at about 7 hours in and 10 hours in with the same error message.  I thought perhaps this is a PI_overheating or throttling issue, so I have added a large fan over the PI and monitored the pi via vcgencmd get_throttled and measure_temp.  Post failure, get_throttled = 0x0 and core temp never exceeded 56C.  

Hardware is a SKR 1.3 board with TMC 2209s running against a PI 3B+.  Serial is connected via a 8 inch cable provided with the SKR 1.3, which looks to be heavy gauge wire and good quality.  PI is a clean build of Buster Raspian.  Monitored top during print and CPU load is negligible at < 10%. 

Log rolled over at midnight in this print, but is attached.  Stats prior to failure:

```
Stats 38592.8: gcodein=19760882 mcu: mcu_awake=0.016 mcu_task_avg=0.000020 mcu_task_stddev=0.000022 bytes_write=74069841 bytes_read=9561177 bytes_retransmit=9 bytes_invalid=0 send_seq=1267437 receive_seq=1267437 retransmit_seq=11 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=48 stalled_bytes=1420 freq=100002821 heater_bed: target=105 temp=105.0 pwm=0.486 print_time=38547.119 buffer_time=3.933 print_stall=0 extruder: target=250 temp=250.4 pwm=0.470
Stats 38593.8: gcodein=19760882 mcu: mcu_awake=0.016 mcu_task_avg=0.000020 mcu_task_stddev=0.000022 bytes_write=74071417 bytes_read=9561419 bytes_retransmit=9 bytes_invalid=0 send_seq=1267463 receive_seq=1267463 retransmit_seq=11 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=34 stalled_bytes=0 freq=100002823 heater_bed: target=105 temp=105.0 pwm=0.486 print_time=38547.119 buffer_time=2.932 print_stall=0 extruder: target=250 temp=250.4 pwm=0.420
```

Not sure what to look at next.  I would really appreciate any suggestions.  Thanks!

[klippy.zip](https://github.com/KevinOConnor/klipper/files/3520645/klippy.zip)

KevinOConnor, 2019-08-21:
Looks like the host python process was unable to run for ~1.4 seconds and thus couldn't meet its deadlines.  It looks like you've implemented a custom install - make sure there isn't anything else on the machine running with higher priority, that the software isn't running in a container (or similar) that could cause it to be deprioritized, etc.

If you're seeing the same failure at the same time in a print, it may be something in the g-code that is confusing Klipper.  If so, could you attach the g-code file that was running at the time of the fault?

-Kevin

darkernoise, 2019-08-21:
Hi Kevin,

Thank you so much for taking time out of your life to not only write Klipper, but to also support it.  That is a huge and appreciated task!  Yes, I did install according to the [Installation document](https://www.klipper3d.org/Installation.html) via git clone. From there, you see the patch I was trying to test for the TMC2209s as I have not switched back to master yet. 

Though, funny that you mention containers as this PI does have OctoPrint-TFT installed, from source and using Docker.   From there, this PI has : Octoprint, mjpg_streamer, and Popt_Star Bridge (my open source project to integrate sensors installed in my printers into MQTT and HomeAssistant).  Any way, I have since stopped everything other than Klipper, Octoprint, and mjpg_streamer and will re-test my long print again tonight.  For that matter, thanks to your advice, I now have a place to start troubleshooting and now I know a bit more about how to interpret the log files.  Thus, I won't take up any more of your time and I will close this Issue. Thank you again, for everything!

closed 2019-08-21 by darkernoise

KevinOConnor, 2019-08-21:
If you're going to retest, you might want to run the g-code through Klipper in "batch mode" first to see if it causes any issues.  Details are in docs/Debugging.md.

-Kevin

darkernoise, 2019-08-23:
I thought I would update that I have re-tested and have some positive results.  I changed the following: removed Octoprint-TFT and Docker from the PI.  The PI was also on an external USB wireless adapter, which could indeed have caused interrupt issues.  Thus, I switched the PI to a hard-wired Ethernet connection.  With those 2 sets of changes, I have successfully repeated and completed the torture test which had previously failed twice.  With that test successful, I have also run several other very long prints (12 and 18 hours) without any issue.  I feel confident that his confirms the issue cause is not Klipper.  I just cannot say for certain which of those two changes is the root cause.

markwinger, 2020-01-06:
I just got this error.  I had my pi completely lock up several time requiring power off to get going again.  I had just installed 'the spaghetti  detective' so I disabled it and tried again.  Ran a few prints fine.  Then enabled the spaghetti detective again and n now get this error.   Appears to be the problem for me.

deviant-aut, 2020-02-06:
I can confirm that spaghetti detective plugin can cause this error.