Klipper3d/klipper #1900: Problem with four independent z_stepper

dzao, 2019-08-20:
Hello, I have a problem with my four independent z_stepper. They wasn´t move. I have two SKR 1.3 with tmc2130 drivers. On the first board i have X, Y, E0, E1 and on second board I have Z, Z1, Z2, Z3. I don´t have any idea why. 

Here is my log and thank you for your help.

Restarting printer
Start printer at Tue Aug 20 19:13:11 2019 (1566328391.9 3715.6)
===== Config file =====
[mcu]
serial = /dev/serial/by-path/platform-3f980000.usb-usb-0:1.1.3:1.0

[mcu zboard]
serial = /dev/serial/by-path/platform-3f980000.usb-usb-0:1.2:1.0

[printer]
kinematics = cartesian
max_velocity = 500
max_accel = 200
max_z_velocity = 25
max_z_accel = 30

[stepper_x]
step_pin = P2.2
dir_pin = !P2.6
enable_pin = !P2.1
step_distance = .00625
endstop_pin = !P1.29
position_endstop = 0
position_max = 300
homing_retract_dist = 10.0
homing_speed = 20

[tmc2130 stepper_x]
cs_pin = P1.17
spi_software_sclk_pin = P0.4
spi_software_mosi_pin = P4.28
spi_software_miso_pin = P0.5
microsteps = 32
interpolate = True
run_current = 2
hold_current = 1
sense_resistor = 0.110
stealthchop_threshold = 50

[stepper_y]
step_pin = P0.19
dir_pin = P0.20
enable_pin = !P2.8
step_distance = .00625
endstop_pin = !P1.27
position_endstop = 0
position_max = 300
homing_retract_dist = 10.0
homing_speed = 20

[tmc2130 stepper_y]
cs_pin = P1.15
spi_software_sclk_pin = P0.4
spi_software_mosi_pin = P4.28
spi_software_miso_pin = P0.5
microsteps = 32
interpolate = True
run_current = 2
hold_current = 1
sense_resistor = 0.110
stealthchop_threshold = 50

[stepper_z]
step_pin = zboard:P2.2
dir_pin = zboard:P2.6
enable_pin = !zboard:P2.1
[... 400 lines cut ...]
Stats 4469.3: gcodein=710 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=5882 bytes_read=92453 bytes_retransmit=9 bytes_invalid=0 send_seq=906 receive_seq=906 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002116 zboard: mcu_awake=0.000 mcu_task_avg=0.000004 mcu_task_stddev=0.000005 bytes_write=76953 bytes_read=28258 bytes_retransmit=9 bytes_invalid=0 send_seq=2885 receive_seq=2885 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002611 adj=100000569 heater_bed: target=0 temp=26.0 pwm=0.000 print_time=719.079 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.0 pwm=0.000 extruder: target=0 temp=26.0 pwm=0.000
Stats 4470.3: gcodein=710 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=5888 bytes_read=92559 bytes_retransmit=9 bytes_invalid=0 send_seq=907 receive_seq=907 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002116 zboard: mcu_awake=0.000 mcu_task_avg=0.000004 mcu_task_stddev=0.000005 bytes_write=76959 bytes_read=28274 bytes_retransmit=9 bytes_invalid=0 send_seq=2886 receive_seq=2886 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002611 adj=100000555 heater_bed: target=0 temp=26.0 pwm=0.000 print_time=719.079 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.0 pwm=0.000 extruder: target=0 temp=26.0 pwm=0.000
Stats 4471.3: gcodein=710 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=5894 bytes_read=92679 bytes_retransmit=9 bytes_invalid=0 send_seq=908 receive_seq=908 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002115 zboard: mcu_awake=0.000 mcu_task_avg=0.000004 mcu_task_stddev=0.000005 bytes_write=76965 bytes_read=28303 bytes_retransmit=9 bytes_invalid=0 send_seq=2887 receive_seq=2887 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002611 adj=100000534 heater_bed: target=0 temp=26.0 pwm=0.000 print_time=719.079 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.0 pwm=0.000 extruder: target=0 temp=26.0 pwm=0.000
Stats 4472.3: gcodein=715 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=5900 bytes_read=92815 bytes_retransmit=9 bytes_invalid=0 send_seq=909 receive_seq=909 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002115 zboard: mcu_awake=0.000 mcu_task_avg=0.000004 mcu_task_stddev=0.000005 bytes_write=76971 bytes_read=28319 bytes_retransmit=9 bytes_invalid=0 send_seq=2888 receive_seq=2888 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002610 adj=100000535 heater_bed: target=0 temp=26.0 pwm=0.000 print_time=719.079 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.0 pwm=0.000 extruder: target=0 temp=26.0 pwm=0.000
Stats 4473.3: gcodein=715 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=5906 bytes_read=92921 bytes_retransmit=9 bytes_invalid=0 send_seq=910 receive_seq=910 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002115 zboard: mcu_awake=0.000 mcu_task_avg=0.000004 mcu_task_stddev=0.000005 bytes_write=76977 bytes_read=28335 bytes_retransmit=9 bytes_invalid=0 send_seq=2889 receive_seq=2889 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002610 adj=100000522 heater_bed: target=0 temp=26.0 pwm=0.000 print_time=719.079 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.0 pwm=0.000 extruder: target=0 temp=26.0 pwm=0.000
Stats 4474.3: gcodein=715 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=5912 bytes_read=93027 bytes_retransmit=9 bytes_invalid=0 send_seq=911 receive_seq=911 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002114 zboard: mcu_awake=0.000 mcu_task_avg=0.000004 mcu_task_stddev=0.000005 bytes_write=76983 bytes_read=28351 bytes_retransmit=9 bytes_invalid=0 send_seq=2890 receive_seq=2890 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002614 adj=100000519 heater_bed: target=0 temp=26.0 pwm=0.000 print_time=719.079 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.0 pwm=0.000 extruder: target=0 temp=26.0 pwm=0.000
Stats 4475.3: gcodein=715 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=5918 bytes_read=93163 bytes_retransmit=9 bytes_invalid=0 send_seq=912 receive_seq=912 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002114 zboard: mcu_awake=0.000 mcu_task_avg=0.000004 mcu_task_stddev=0.000005 bytes_write=76989 bytes_read=28367 bytes_retransmit=9 bytes_invalid=0 send_seq=2891 receive_seq=2891 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002613 adj=100000577 heater_bed: target=0 temp=26.0 pwm=0.000 print_time=719.079 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.0 pwm=0.000 extruder: target=0 temp=26.0 pwm=0.000
Stats 4476.3: gcodein=715 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=5924 bytes_read=93283 bytes_retransmit=9 bytes_invalid=0 send_seq=913 receive_seq=913 retransmit_seq=2 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002114 zboard: mcu_awake=0.000 mcu_task_avg=0.000004 mcu_task_stddev=0.000005 bytes_write=76995 bytes_read=28396 bytes_retransmit=9 bytes_invalid=0 send_seq=2892 receive_seq=2892 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002614 adj=100000554 heater_bed: target=0 temp=26.0 pwm=0.000 print_time=719.079 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.0 pwm=0.000 extruder: target=0 temp=26.0 pwm=0.000
Stats 4477.3: gcodein=720 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=5930 bytes_read=93389 bytes_retransmit=9 bytes_invalid=0 send_seq=914 receive_seq=914 retransmit_seq=2 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002114 zboard: mcu_awake=0.000 mcu_task_avg=0.000004 mcu_task_stddev=0.000005 bytes_write=77001 bytes_read=28412 bytes_retransmit=9 bytes_invalid=0 send_seq=2893 receive_seq=2893 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002614 adj=100000552 heater_bed: target=0 temp=26.0 pwm=0.000 print_time=719.079 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.0 pwm=0.000 extruder: target=0 temp=26.0 pwm=0.000
Stats 4478.3: gcodein=720 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=5936 bytes_read=93525 bytes_retransmit=9 bytes_invalid=0 send_seq=915 receive_seq=915 retransmit_seq=2 srtt=0.001 rt [... cut ...] 025 ready_bytes=0 stalled_bytes=0 freq=100002113 zboard: mcu_awake=0.000 mcu_task_avg=0.000004 mcu_task_stddev=0.000005 bytes_write=77097 bytes_read=28704 bytes_retransmit=9 bytes_invalid=0 send_seq=2909 receive_seq=2909 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002609 adj=100000481 heater_bed: target=0 temp=26.0 pwm=0.000 print_time=719.079 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.0 pwm=0.000 extruder: target=0 temp=26.0 pwm=0.000

KevinOConnor, 2019-08-23:
If you're having issues with the stepper configuration, make sure you run through the checks at https://www.klipper3d.org/Config_checks.html .

-Kevin

dzao, 2019-08-25:
Hello, I can´t see any problem with my config. Bltouch on my zboard is working, axes x,y on my main board working, temps showing. If I send dump_tmc stepper= , looks like everything working, but they won´t move. Can you have problem in firmware with two lpc1768 cpu??

BlackStump, 2019-08-26:
have you tested the boards independently to make sure they are functioning?

dzao, 2019-08-26:
Yes, i have some output on this board and bltouch is working too

dzao, 2019-09-09:
It has be a problem with klipper. Now I´m runing with DRV8825 without SPI  and everything works fine. It will be a problem in SPI communication with two boards and with drivers on both (TMC2130).

KevinOConnor, 2019-09-14:
I'm not sure what the status of this issue is.  If you think there is an issue with the software then please provide a detailed description of your efforts running the steps at https://www.klipper3d.org/Config_checks.html (what worked, what didn't work) along with a log (including M112) after running any step that did not work correctly.

-Kevin

closed 2019-09-29 by KevinOConnor