Klipper3d/klipper #1901: Display temperature

iamacell, 2019-08-20:
Hello!

Would it be possible to modify the display.py to display the temperatures a bit different? 
I mean, not only 30°, but 30°C. 
from this line: self.draw_text(x,   y, "%3.0f~degrees~" % (temperature,))
to this: self.draw_text(x, y, "%3.0f~degrees~C" % (temperature,))

and while reaching target temp (as there is not enough space on display to add "C") only eg 30 -> 180
this line: self.draw_text(x, y, "%3.0f~right_arrow~%.0f~degrees~" % (
to this: self.draw_text(x, y, "%3.0f~right_arrow~%.0f" % (

Also would it be possible to add a variable for the printer name? So not just display "ready", but "XY printer ready"?

Thank you!

KevinOConnor, 2019-08-23:
It's possible to customize the display - there are some details in config/example-menu.cfg.  However, note that the menu is currently in flux as we change it to use templates.

-Kevin

iamacell, 2019-08-24:
Hello Kevin!

Yes, I can modify the menu (order, items, add comands, etc...) but I cannot modify the main info screen (with the temperatures, etc) from where you can go to the menu. Whatever I do I can see the default only. 
There are only 20x4 and 16x4 screens in the menu.cfg. I have reprap discount 12864 (lcd_type: st_7920).

Whatever I modify in this (or similar) section

_[menu __card2004_static]
type: card
name: Card 20x04
content:
    "\x00{0:3s}\x04\x7e{1:3s}\x04   {2}"
    "\x01{3:3s}\x04\x7e{4:3s}\x04   \xa5{8}"
    "\x07{6}  \x02{5} {7}"
    "{9}"
items:
    __card_hotend0_current
    __card_hotend0_target
    __card_zpos
    __card_hotbed_current
    __card_hotbed_target
    __card_frpeed
    __card_fnspeed
    5,__card_prt_time, __card_usb_progress, __card_sdcard_progress
    __card_status
    __card_msg,__card_xpos|__card_ypos|__card_epos_

takes no effect.

KevinOConnor, 2019-09-06:
Alas, the best documentation we have is in config/example-menu.cfg.  As noted above, the menu is currently in flux, so this will likely change in the coming weeks/months.

-Kevin

iamacell, 2019-09-06:
Great! Thank you!

KevinOConnor, 2019-09-14:
Okay, I'm going to close this as I think the original question has been answered (at least as best as we could).

-Kevin

closed 2019-09-14 by KevinOConnor