Klipper3d/klipper #1903: ENDSTOP trigger during print

CopterFail, 2019-08-22:
Question: 
How does Klipper behave on temporary disturbance (triggered by EMF, cable, LED light) of an ENDSWITCH? 
Is the stepper stopped but the internal position updated, resulted in a layer shift (permanent for this print)?

KevinOConnor, 2019-08-23:
Klipper only checks the endstops during homing.  During homing, there is a filter on the trigger check (the pin is sampled 4 times over ~60us to ensure it really is triggered).

-Kevin

CopterFail, 2019-08-24:
Good solution, thank you

closed 2019-08-24 by CopterFail

webers89, 2020-12-01:
I found this discussion is similar to my question. 
Is there option to enable that Klipper checks endstops all the time not only during homing process? 
I want this to be as a additional safety if stepper motors loses steps.