Klipper3d/klipper #1906: Attempting to use virtual endstop as probe gives internal error

bryanboettcher, 2019-08-23:
[tmc2130 stepper_z] 
diag1_pin: !PC4 

[stepper_z]
#endstop_pin: tmc2130_stepper_z:virtual_endstop
endstop_pin: probe:z_virtual_endstop
position_endstop: 0
homing_retract_dist: 0

[probe]
pin: tmc2130_stepper_z:virtual_endstop
#pin: !PC4
z_offset: 0

[klippy.log](https://github.com/KevinOConnor/klipper/files/3534409/klippy.log)

I stripped out the not-relevant bits of the config when pasting here (although full config is in the log).  Attempting to use the Z virtual endstop as a probe configuration fails immediately on startup with an error instead of a config check.  I would expect, at a minimum, this to be caught and reported with something like "Klipper will not use Z virtual endstop as a probe", although IMO that should be up to me as the printer owner.

Can we make the printer operator opt-in to this behavior, with something like:

[probe]
pin: tmc2130_stepper_z:virtual_endstop
*enable_dangerous_probe: True*
z_offset: 0

AndyWo83, 2019-08-23:
Hello, i hope I can help you...
I have these configuration and it works.

[tmc2130 stepper_z]
diag1_pin: !PC4 ### <-- NO

[stepper_z]
endstop_pin: probe:z_virtual_endstop
#position_endstop: 0.1 ### <-- Must comment out
position_min: -2 ### <-- Must a negative Number
position_max: 240

[probe]
pin: ^ar18 ### <-- MKS GenL Pin, use yours
z_offset = 0.880

bryanboettcher, 2019-09-04:
It's not a matter of picking the right pins or values, it's that the error verbiage is inconsistent.

KevinOConnor, 2019-09-06:
I agree that Klipper could have done a better job of reporting an error.  I didn't see a simple fix for that, however.

As for using "stallguard" to probe - a developer would need to implement the code to make that work.  The current sensorless homing implementation (as the virtual pin name and documentation indicate) is specific to homing.

-Kevin

jakep82, 2019-09-06:
I think it would be a mistake to add support for stallguard as a probe method.  It will create a maintenance burden when people inevitably report issues because they receive inaccurate and inconsistent results.  Just my 2 cents.

KevinOConnor, 2019-11-06:
I'm going to close this as it looks like the conversation has concluded.  I looked again at the error reporting and don't see a simple way to improve the error message.

-Kevin

closed 2019-11-06 by KevinOConnor