Klipper3d/klipper #1907: lowered endstops and raised the bed - re-calibrate

peterlpop, 2019-08-23:
I have had the delta calibrate and the Enhanced done - worked like a charm. 

I now raised the bed and lowered the endstops - 32mm total. 

I tried probe offset, manual and probe calibration. Read and re-read all calibration docs. Short of redoing the whole config file... 

I can't figure out how to tell the printer it's 580mm tall now, instead of 612mm. It's a k280 Delta. 

Every time I try to calibrate, it wants to go under the bed 30+ mm - probe or manual. 
Please help. Thanks.

peterlpop, 2019-08-24:
I changed the "position_endstop =" to current less 32mm ... in the "DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated." - this allowed Delta_calibrate to start before hitting the bed. 

now I can run another enhanced calibrate. 

If there is a better way to make it "hunt" for the bed earlier/higher, I think it would help greatly - especially with the initial setup.

Hywelmartin, 2019-08-26:
for delta calibrate you set the initial height [here..](https://github.com/KevinOConnor/klipper/blob/8f5c15d4dc6a6f592a09e1d5ed80516d5deb8100/config/example-delta.cfg#L123)

closed 2019-08-30 by klipper-gitissuebot